jaheyns/CfdOF #11: Import self-created structured mesh-file

dominikstaltner, 2018-12-10:
Hello guys,
is it possible to import a self-created structured mesh-file (e.g. .stl-file created in gmsh) to the CfdOF-workbench in FreeCAD?
Best regards
Dominik

oliveroxtoby, 2018-12-11:
Hi Dominik

It is possible in principle - you can open the stl file in FreeCAD and use it as a surface the same way you would use a surface that you created in FreeCAD. However, it is not very practical if the surface contains even a moderate number of triangles, as each will be seen as a separate face when assigning boundary conditions and the process will quickly become impractical and/or slow.

Doing this in an elegant way is probably the most wanted feature of CfdOF right now. Personally I would like to see a way of grouping the faces of the imported stl into some sort of composite faces.

If you or anyone else has an interest in looking at this, it would be great.

dominikstaltner, 2018-12-12:
Hello Oliver,

thanks a lot for your answer. It would be really cool if this will work in the future (like the import of .unv-files in the FEM-workbench).

Best regards
Dominik

commit eecb9c63 (2021-08-05): Allow selection of whole objects

To improve performance and practicality in selecting
boundary conditions etc for many-faced objects, particularly
imported triangulated geoemetries such as .stl and .obj

Resolves #11
closed 2021-08-05 by oliveroxtoby