jaheyns/CfdOF #12: HISA not found even installation attempt

Titi9237, 2019-01-15:
Hi, 
I am trying to install HISA with the Freecad Preferences assistant and it is not working for me : 
[image]

When using the Install HISA button, I end up with 
> Checking dependencies...
HiSA not found
Downloading HiSA, please wait...
Extracting HiSA...
Download completed
Building HiSA. Please wait...
Log file: /home/myusername /OpenFOAM/myusername-6/hisa-master/log.Allwmake
Install completed
Checking dependencies...
HiSA not found

The log.Allwmake yields many errors (see in attachments) 
[log.Allwmake.txt](https://github.com/jaheyns/CfdOF/files/2760092/log.Allwmake.txt)

I am using Ubuntu 18.04 and Freecad 0.17.

What am I doing wrong ? If you need any other info, feel free to ask.

Thanks

oliveroxtoby, 2019-01-15:
It seems that I forgot to update the package of HiSA with the latest one updated for OF6. If you retry the installation now, is should be working.

closed 2019-01-15 by oliveroxtoby

Titi9237, 2019-01-16:
Thanks, It seems to be fixed !

Estay1991, 2020-06-24:
I encountered same problem. Hisa is not found after installation complete.

oliveroxtoby, 2020-06-24:
> I encountered same problem. Hisa is not found after installation complete.

Please could you log a new issue, providing full information about how to reproduce your problem.

Estay1991, 2020-06-24:
just logged a new issue.  https://github.com/jaheyns/CfdOF/issues/35
Please help.