jaheyns/CfdOF #15: MacOSx

fpiergen, 2019-02-24:
Any plans to put in instructions to get this working with MAC? Can't figure out how to integrate in the Docker container to have openFoam installation available to Freecad on MAC

oliveroxtoby, 2019-02-25:
We would need someone to contribute this, as the developers don't have access to Mac.

If it is an issue with OpenFOAM installation, there should be information out there, as there are a huge number of OpenFOAM users.

olesenm, 2019-02-27:
As @oliveroxtoby points out we rely on external enthusiasts for OSX testing. I thus don't know if there are any osx-related problems in the last version, but OpenFOAM -v1806 added native OSX support. By native I mean compilation from source.

See the portability section:
https://www.openfoam.com/releases/openfoam-v1806/build.php#build-configuration

fpiergen, 2019-03-01:
I will give that a try and report back. Thank you for information.

fpiergen, 2019-04-01:
Got this compiled on MAC and almost working. When I try to run a set up case CfDof is looking for a python module that does not exist in Freecad.

`matplotlib not found, so Plot module can not be loaded
Traceback (most recent call last):
  File "/Users/fabio/Library/Preferences/FreeCAD/Mod/CfdOF/_ViewProviderCfdSolverFoam.py", line 81, in setEdit
    from CfdRunnableFoam import CfdRunnableFoam
  File "/Users/fabio/Library/Preferences/FreeCAD/Mod/CfdOF/CfdRunnableFoam.py", line 44, in <module>
    from CfdResidualPlot import ResidualPlot
  File "/Users/fabio/Library/Preferences/FreeCAD/Mod/CfdOF/CfdResidualPlot.py", line 30, in <module>
    from freecad.plot import Plot
<type 'exceptions.ImportError'>: No module named plot`

Having a hard time trying to import the python module matplotlib  into Freecad.
Any suggestions?
Thanks.

oliveroxtoby, 2019-04-01:
Good to hear! 

On my Linux machine I recall having to install matplotlib through my system package manager. (It wasn't sufficient just to install the python module using e.g. pip.)

I think this should be as much of an issue with the Plot module as with CfdOF, and that presumably has been tested on Mac. [This thread](https://forum.freecadweb.org/viewtopic.php?f=22&t=27013&start=20) suggests matplotlib should be included in the AppImage -  are you using it or compiling from source?

fpiergen, 2019-04-01:
No I am using version 0.17 latest free cad.  I was able to get  matplotlib module but now I get a different error.
`Traceback (most recent call last):
  File "/Users/fabio/Library/Preferences/FreeCAD/Mod/CfdOF/_ViewProviderCfdSolverFoam.py", line 81, in setEdit
    from CfdRunnableFoam import CfdRunnableFoam
  File "/Users/fabio/Library/Preferences/FreeCAD/Mod/CfdOF/CfdRunnableFoam.py", line 44, in <module>
    from CfdResidualPlot import ResidualPlot
  File "/Users/fabio/Library/Preferences/FreeCAD/Mod/CfdOF/CfdResidualPlot.py", line 28, in <module>
    import Plot
  File "/Applications/FreeCAD.app/Contents/Mod/Plot/Plot.py", line 32, in <module>
    matplotlib.use('Qt4Agg')
<type 'exceptions.AttributeError'>: 'module' object has no attribute 'use'`

Progress...:)

fpiergen, 2019-04-01:
Ok I got passed  that. I was using the wrong version . Need to use 2.7 python. Will keep you posted.

fpiergen, 2019-04-02:
This is what I used as a guide to compile on MAC
[https://github.com/mrklein/openfoam-os-x/wiki/OpenFOAM(R)-git-version-&-Homebrew](url)

oliveroxtoby, 2019-04-02:
Thank you for your efforts! Does it seem to be working now? I will add this info to the Readme.

fpiergen, 2019-04-02:
I have not tried to run the case yet. On thing I did notice is that it had no place to input where I want to write the files for openfoam. It used a temporary directory. /var/folders/k5/srn0g_vn43sfnk93p1pwv3140000gn/T/case

I was able to write the case to that area. When I have some more time I will do the run part and see if I can view the results.

Thanks,
Fabio

oliveroxtoby, 2019-04-02:
You should be able to set this in the preferences dialog - Edit | Preferences | CFD, Default output directory.

fpiergen, 2019-04-02:
Got it thanks. The tutorial I was following showed it in the run dialog of the CfDof workbench. I thought it was a bug as I did not have that option. Maybe different on different platforms.

fpiergen, 2019-04-02:
This is is what I did in Freecad python console to get the matplotlib imported:
`import sys 
 sys.path.append('/System/Library/Frameworks/Python.framework/Versions/2.7/Extras/lib/python’)
 import matplotlib`

oliveroxtoby, 2019-04-10:
So it turns out that the Plot workbench has been moved out of the FreeCAD core into an external module. Some packages include it (and its dependency, matplotlib), others don't. In future it will be installable from the AddOn manager, but for now a workaround like this is necessary. I have updated the Readme and dependency checker accordingly. For now it's a bit of a pain unfortunately.

closed 2019-04-10 by oliveroxtoby

fpiergen, 2019-04-28:
Went through elbow tutorial in 
http://www.opensim.co.za/training.html

Everything seems to be working. A little buggy but it works. Thanks for help.