jaheyns/CfdOF #22: Mesh creation errors

tomkcook, 2019-06-30:
I have a custom workbench that creates convex faces (`TopoDS_Face`) that represent the shapes of yacht hulls.  Unfortunately it's probably not feasible for you to install the workbench or load files it creates at this point.  The result looks like this:

[image]

I'd now like to simulate this in a stream of water.  The water surface should be around half-way up the side of the hull.  I want to include surface wave simulation and so I should need two initial regions of fluid, one for the water & one for the air.  I've set these up as cubes:

[image]

Now I'm trying to generate a mesh for this simulation.  It doesn't appear to be possible to create a mesh for multiple objects, as the 'Create a mesh...' button on the toolbar is only enabled when one object is selected.  I've tried using the Part workbench's 'Create compound' tool to combine the objects into a single one, but then mesh generation produces this output:

```
0.0: Preparing meshing ...
0.0: Error: 'NoneType' object has no attribute 'Name'
```

What's the recommended way forward here?

oliveroxtoby, 2019-07-01:
It's intended that you should have one mesh for the entire geometry, and then create an 'initialisation zone' to specify the portion initially filled with water. I'll have a look at the compound issue though.

oliveroxtoby, 2019-07-01:
I'm not able to reproduce the meshing issue, so I would have to ask for more information (please see the readme for bug reporting guidelines).

tomkcook, 2019-07-01:
Here is the Report View output:

```
Starting meshing ...

  Part to mesh: Name --> Compound,  Label --> Compound, ShapeType --> Compound
  CharacteristicLengthMax: 690.0
  ElementDimension: 3D
Traceback (most recent call last):
  File "/home/tkcook/.FreeCAD/Mod/CfdOF/_TaskPanelCfdMesh.py", line 252, in writeMesh
    FreeCADGui.doCommand("cart_mesh.get_file_paths(CfdTools.getOutputPath(FreeCAD.ActiveDocument." + self.analysis.Name + "))")
AttributeError: 'NoneType' object has no attribute 'Name'
```

I've attached an fcstd file that I hope should open in stock FreeCAD that demonstrates the problem.

Here's the version info - note that I've built from source, the version is commit 3132dfbf542ff54af4a6e688422373690d4c57f8:
```
OS: Ubuntu 18.04.2 LTS (GNOME/gnome-xorg)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.17178 (Git)
Build type: Debug
Branch: master
Hash: 3132dfbf542ff54af4a6e688422373690d4c57f8
Python version: 2.7.15rc1
Qt version: 4.8.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedKingdom (en_GB)
```

[test.zip](https://github.com/jaheyns/CfdOF/files/3345232/test.zip)

tomkcook, 2019-07-01:
Never mind, got it.  This is a silly bug, not a deep one; you shouldn't enable the 'Create a mesh...' button until an analysis object has been created!

closed 2019-07-01 by tomkcook

oliveroxtoby, 2019-07-01:
OK, great. BTW I think this issue (enabling the mesh button without an analysis object) has been taken care of in the latest update.
Thanks for your interest :-)