jaheyns/CfdOF #23: Materials error

naikymen, 2019-07-11:
Hi, I am running FreeCAD from the latest AppImage.
```
OS: Arch Linux
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.18.16117 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.18.2)
Hash: dbb4cc6415bac848a294f03b80f65e888d531742
Python version: 3.6.7
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

When I press "Add fluid properties" the following error occurs:
```
File CardName (SeaWater) is not content CardName (FreeCAD Material card: see https://www.freecadweb.org/wiki/Material)
Running the Python command 'Cfd_Analysis' failed:
Traceback (most recent call last):
  File "/home/nicomic/.FreeCAD/Mod/CfdOF/CfdAnalysis.py", line 94, in Activated
    FreeCADGui.doCommand("analysis.addObject(CfdFluidMaterial.makeCfdFluidMaterial('FluidProperties'))")
  File "<string>", line 1, in <module>
  File "/home/nicomic/.FreeCAD/Mod/CfdOF/CfdFluidMaterial.py", line 39, in makeCfdFluidMaterial
    _CfdMaterial(obj)  # Include default fluid properties
  File "/home/nicomic/.FreeCAD/Mod/CfdOF/CfdFluidMaterial.py", line 91, in __init__
    self.initProperties(obj)
  File "/home/nicomic/.FreeCAD/Mod/CfdOF/CfdFluidMaterial.py", line 104, in initProperties
    mats, name_path_list = CfdTools.importMaterials()
  File "/home/nicomic/.FreeCAD/Mod/CfdOF/CfdTools.py", line 1129, in importMaterials
    material_name_path_list = material_name_path_list + addMatDir(system_mat_dir, materials)
  File "/home/nicomic/.FreeCAD/Mod/CfdOF/CfdTools.py", line 1143, in addMatDir
    materials[a_path] = importFCMat.read(a_path)
  File "/tmp/.mount_FreeCAcAb56k/usr/Mod/Material/importFCMat.py", line 122, in read
    v = line.split(";")[1].strip()  # Line 2
```

And I cannot select any materials in FreeCAD.

Original materials, produce the error because they are missing a comment in the second line:
```
; FreeCAD Material card: see https://www.freecadweb.org/wiki/Material

[FCdMat]
Name = Water
Description = Standard distilled water properties at 20 Degrees Celsius and 1 atm
Density = 998 kg/m^3
DynamicViscosity = 1.003e-3 kg/m/s 
```

Materials with second line comment do not:
```
; FreeCAD Material card: see https://www.freecadweb.org/wiki/Material
; FreeCAD
[FCdMat]
Name = Water
Description = Standard distilled water properties at 20 Degrees Celsius and 1 atm
Density = 998 kg/m^3
DynamicViscosity = 1.003e-3 kg/m/s 
```

It seems that missing "; AUTHOR LICENCE STUFF" in the second line was expected by the code in `importFCMat.py`, which is missing in every material for some reason.

Hope it helps.

luzpaz, 2019-07-11:
@naikymen care to test with the [0.19dev AppImage](https://github.com/FreeCAD/FreeCAD/releases/tag/0.19_pre) ?

naikymen, 2019-07-11:
Yep, seems to work in that one.

```
OS: Arch Linux (KDE)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.17331 (Git) AppImage
Build type: Release
Branch: master
Hash: cf18161dc26267c259ac21675cb9637263bf4152
Python version: 3.7.3
Qt version: 5.9.7
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: English/UnitedStates (en_US)
```

Thanks!

luzpaz, 2019-07-11:
Great to hear. Please close this ticket then. Thanks.

oliveroxtoby, 2019-07-11:
It ought to be backward compatible with 0.18. In fact I thought I had tested this, but clearly something went awry. Thanks for the helpful report!

closed 2019-07-11 by oliveroxtoby

luzpaz, 2019-07-11:
please forgive my presumptuousness

oliveroxtoby, 2019-07-12:
No worries! Appreciate the input :-)

commit 96f77f50 (2019-07-11): Make material cards properly backward compatible with 0.18

Resolves #23