jaheyns/CfdOF #24: InitialiseFields break

fb-multimedia, 2019-09-14:
Hello,
First I would thank you for all your effort in this project.
I'm beginner in Freecad and CFD.
 I've try make a very simple case with cube and a cylindre in but when I clic "InitialiseFields" button This error appear :
`Traceback (most recent call last):
  File "/home/fred/.FreeCAD/Mod/CfdOF/CfdInitialiseFlowField.py", line 149, in setEdit
    self.Object, physics_model, boundaries, material_objs)
  File "/home/fred/.FreeCAD/Mod/CfdOF/_TaskPanelCfdInitialiseInternalFlowField.py", line 62, in __init__
    self.load()
  File "/home/fred/.FreeCAD/Mod/CfdOF/_TaskPanelCfdInitialiseInternalFlowField.py", line 104, in load
    self.form.comboInlets.setCurrentIndex(self.form.comboInlets.findData(self.obj.Inlet.Name))
<class 'AttributeError'>: 'NoneType' object has no attribute 'Name'
`
and the task dialog does'nt open.
Thanks for your help.

oliveroxtoby, 2019-09-15:
Oops! Thank you for reporting. Should be fixed.

closed 2019-09-15 by oliveroxtoby

fb-multimedia, 2019-09-15:
Thanks a lot !

commit 3f560d7a (2019-09-15): Fix big introduced in previos commit when inlet is None in initialisation

Fixes #24