jaheyns/CfdOF #25: AttributeError: module 'Plot' has no attribute 'figure'

fb-multimedia, 2019-09-15:
Case write is ok but when I'm running the solver I got this error :
`Start to write case to folder /tmp
Matching boundary conditions ...
Populating createPatchDict to update BC names
Successfully wrote case to folder /tmp
Traceback (most recent call last):
  File "/home/fred/.FreeCAD/Mod/CfdOF/_TaskPanelCfdSolverControl.py", line 152, in runSolverProcess
    cmd = self.solver_runner.get_solver_cmd(solverDirectory)
  File "/home/fred/.FreeCAD/Mod/CfdOF/CfdRunnableFoam.py", line 93, in get_solver_cmd
    self.residualPlot = ResidualPlot()
  File "/home/fred/.FreeCAD/Mod/CfdOF/CfdResidualPlot.py", line 37, in __init__
    self.fig = Plot.figure(FreeCAD.ActiveDocument.Name + "Residuals")
AttributeError: module 'Plot' has no attribute 'figure'`

Is it a python3, plot module or cfdOF bug ?

oliveroxtoby, 2019-09-15:
I can't reproduce this on my (non-Ubuntu) system. I am wondering: did you install the Plot module through Tools | Modules or does it come packaged with FreeCAD when you install the ppa?

fb-multimedia, 2019-09-15:
Thanks for your help
I've install it with tools/modules

fb-multimedia, 2019-09-15:
Installing freecad v0.19 solve the issue. It's work now !
Thanks for your help

closed 2019-09-15 by fb-multimedia