jaheyns/CfdOF #29: Running the Python command 'Cfd_Analysis' failed

oddtopus, 2019-12-21:
Hi,
I'm not able even to create an analysis because.
It seems an error similar to issue #23 but it happens immediately when I create the analysis. 
Namely: the container CfdAnalysis is created in the document and has also PhysicsModel inside it Instead FluidProperties is left outside  the container, in the document tree, and then the script stops.
I tried to update the addon cfdof but the addon manager returns that it is already up-to-date.
The full error message is:
'''
Running the Python command 'Cfd_Analysis' failed:
Traceback (most recent call last):
  File "/home/user/.FreeCAD/Mod/CfdOF/CfdAnalysis.py", line 94, in Activated
FreeCADGui.doCommand("analysis.addObject(CfdFluidMaterial.makeCfdFluidMaterial('FluidProperties'))")
  File "<string>", line 1, in <module>
  File "/home/user/.FreeCAD/Mod/CfdOF/CfdFluidMaterial.py", line 39, in makeCfdFluidMaterial
	_CfdMaterial(obj)  # Include default fluid properties
  File "/home/user/.FreeCAD/Mod/CfdOF/CfdFluidMaterial.py", line 91, in init
	self.initProperties(obj)
  File "/home/user/.FreeCAD/Mod/CfdOF/CfdFluidMaterial.py", line 104, in initProperties
	mats, name_path_list = CfdTools.importMaterials()
  File "/home/user/.FreeCAD/Mod/CfdOF/CfdTools.py", line 1150, in importMaterials
	material_name_path_list = material_name_path_list + addMatDir(system_mat_dir, materials)
  File "/home/user/.FreeCAD/Mod/CfdOF/CfdTools.py", line 1166, in addMatDir
	materials[a_path] = importFCMat.read(a_path)
  File "/tmp/.mount_FreeCANAkZYR/usr/Mod/Material/importFCMat.py", line 115, in read
	v = line.split(";")[1].strip()  # Line 1

list index out of range
'''
I'm running 0.19 dev Appimage but that happens also with 0.18 Appimage and 0.17:
'''
OS: Linux Mint 19.2
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.16121 (Git) AppImage
Build type: Release
Branch: master
Hash: a7919b63dea4bb1f1107c80ed44758479eec3d8d
Python version: 3.6.7
Qt version: 5.6.2
Coin version: 4.0.0a
OCC version: 7.3.0
Locale: Italian/Italy (it_IT)
'''
Can you clarify how to get rid of this?

oliveroxtoby, 2019-12-21:
This error is occurring when workbench attempts to read the contents of any .FCMat files in the <module directory>/data/CfdFluidMaterialProperties directory - usually ~/.FreeCAD/Mod/CfdOF/data/CfdFluidMaterialProperties

In order for me to try to reproduce the error, please could you attach any .FCMat files you have in this directory on your system?

oddtopus, 2019-12-22:
Goodmorning and thanks for the prompt reply.
The files in the directory are:
Air.FCMat  Gasoline.FCMat  None.FCMat  OilSAE10W.FCMat	Readme.txt  SeaWater.FCMat  Water.FCMat
The typical content is:
; Air
;
; FreeCAD Material card: see https://www.freecadweb.org/wiki/Material

[FCMat]
Name = Air
Description = Standard air properties at 20 Degrees Celsius and 1 atm
Density = 1.20 kg/m^3
DynamicViscosity = 1.80e-5 kg/m/s
MolarMass = 0.0289643897748887 kg/mol
Cp = 1004.703 J/kg/K
SutherlandTemperature = 110.4 K
SutherlandRefViscosity = 1.716e-5 kg/m/s
SutherlandRefTemperature = 273.15 K

oliveroxtoby, 2019-12-22:
"None.FCMat" should not be there. Please delete it and try again. I presume it's a left-over from an old version of the workbench.

oddtopus, 2019-12-25:
Now everything works fine!
Thanks for the support.

closed 2019-12-25 by oddtopus