jaheyns/CfdOF #31: blueCFD automatic download fails to launch installer

alafr, 2020-03-26:
Hi,
If i click install blueCFD-core in the preferences dialog, it downloads a file with no extension instead of .exe and it fails to launch the installer
[image]
[image]
Translation: There is no association of an application with the specified file for this operation
[image]
[image]

If i download the file by copying the given url https://github.com/blueCFD/Core/releases/download/blueCFD-Core-2017-2/blueCFD-Core-2017-2-win64-setup.exe in the browser, and then select it with choose existing file, it works fine.

oliveroxtoby, 2020-03-26:
Thanks for reporting. Should be fixed.

closed 2020-03-26 by oliveroxtoby

alafr, 2020-03-26:
Yes it's fixed. Thanks!

commit f9b43a2c (2020-03-26): Fix BlueCFD download to use .exe extension

Fixes #31