jaheyns/CfdOF #32: OpenFOAM IO error: Cannot find patchField entry for MeshRefinement... file: ../case/0/U.boundaryField from line 28 to line 19.

joseRey, 2020-04-04:
Somehow, when trying to run a case that has a mesh with a "zero thickness baffle" defined in a surface mesh refinement, I always get the following solver error:

OpenFOAM IO error: Cannot find patchField entry for MeshRefinement... file: ../case/0/U.boundaryField from line 28 to line 19.
(removed path)

This happens pretty consistently whenever I use the baffle option for surface refinement for snappyHexMesh.

Whether I define the baffle surfaces as a no-slip wall or just not define them at all for the solver, I get the same error.

Awesome work.  Thank you so much.

oliveroxtoby, 2020-04-05:
Could you upload a simple case to reproduce this? It seems to work ok on the training example at http://www.opensim.co.za/training.html

joseRey, 2020-04-05:
I uploaded a model similar to the baffle model in the examples.  The difference is that I am trying to make the baffle a wall boundary.  Please see attachment here.

[modifiedBaffleRectangularTube.zip](https://github.com/jaheyns/CfdOF/files/4434444/modifiedBaffleRectangularTube.zip)

joseRey, 2020-04-12:
I found that this is not an issue but would rather be an enhancement since cfdOF would have to perform what is done in [this youtube tutorial (to make the baffle into a wall)](https://www.youtube.com/watch?v=7Ex1RHDQqIc&t=308s).

However, there is a workaround (partially learned this from the tutorials).  The workaround is to make all the walls into faces (exploding with the draft workbench) and then creating a compound with the parts workbench.  In cases of more complex geometries, XOR can precede the exploding.  The mesher nor the solver care that the compound is a shell and not a solid.

FreeCad file; it has everything required (note it is set to mesh and solve in 4 "cores"):
[modifiedBaffleRectangularTube+XOR+Explode+Compound.zip](https://github.com/jaheyns/CfdOF/files/4466746/modifiedBaffleRectangularTube%2BXOR%2BExplode%2BCompound.zip)

The model in the FreeCad file:
[image]

Paraview visualization (velocity in clip and streamlines):
[image]

Again, Thank you very much for this awesome contribution.  cfdOF is at the intersection of two of my favorite Open Source projects.

oliveroxtoby, 2021-08-09:
Closing this as I think there should no longer be this ambiguity - an baffle can only be a boundary type that is designed as a baffle, otherwise to create a wall, the face(s) in question need to be included in the geometry specification (e.g. by creating a compound) as for other boundaries.

closed 2021-08-09 by oliveroxtoby