jaheyns/CfdOF #33: Installation: error while installing cfmesh in new computer

joseRey, 2020-04-13:
Hi,

I am getting a lot of errors when running the installation of cfmesh using the latest zip file from sourceforge (march 2020).  However, it keeps on looking for directories that do not seem to exist, nor could I find them in the ZIP's folder (I also looked for them in another computer with the installation and could not find them).

One Example from the log file:
could not open file UList.H for source file utilities/workflowControls/workflowControls.C
(I couldn't find this directory anywhere).

Here is a screenshot of the panel:
[image]

***  Here is the initial portion of the log file ***
`wmake libso meshLibrary
wmakeLnInclude: linking include files to ./lnInclude
Making dependency list for source file workflowControls.C
Making dependency list for source file fpmaMesh.C
Making dependency list for source file writeMeshFPMA.C
Making dependency list for source file voronoiMeshGenerator.C
Making dependency list for source file voronoiMeshExtractorCreateMesh.C
Making dependency list for source file voronoiMeshExtractorAddressing.C
Making dependency list for source file voronoiMeshExtractor.C
Making dependency list for source file tetMeshGenerator.C
Making dependency list for source file tetMeshExtractorOctree.C
Making dependency list for source file cartesian2DMeshGenerator.C
Making dependency list for source file cartesianMeshGenerator.C
Making dependency list for source file cartesianMeshExtractorDecomposeSplitHexes.C
could not open file UList.H for source file utilities/workflowControls/workflowControls.C
could not open file foamTime.H for source file utilities/workflowControls/workflowControls.C
could not open file Time.H for source file utilities/workflowControls/workflowControls.C
could not open file HashSet.H for source file utilities/workflowControls/workflowControls.C
could not open file token.H for source file utilities/workflowControls/workflowControls.C
could not open file Istream.H for source file utilities/workflowControls/workflowControls.C
could not open file SubList.H for source file utilities/workflowControls/workflowControls.C
could not open file foamTime.H for source file utilities/dataConversion/foamToFPMA/fpmaMesh.C
could not open file Time.H for source file utilities/dataConversion/foamToFPMA/fpmaMesh.C
could not open file HashSet.H for source file utilities/dataConversion/foamToFPMA/fpmaMesh.C
could not open file token.H for source file utilities/dataConversion/foamToFPMA/fpmaMesh.C
could not open file UList.H for source file utilities/dataConversion/foamToFPMA/fpmaMesh.C
could not open file Istream.H for source file utilities/dataConversion/foamToFPMA/fpmaMesh.C
could not open file SubList.H for source file utilities/dataConversion/foamToFPMA/fpmaMesh.C`
*********

Thank you very much for this wonderful project.

j2l, 2020-04-23:
Great project, thanks!
I also have install errors, on Ubuntu for me:
[image]

j2l, 2020-04-23:
After `chown -R pm openfoam7/` I have new errors:
```
/bin/bash: /tmp/.mount_FreeCAuXtklB/usr/lib/libtinfo.so.6: no version information available (required by /bin/bash)
/bin/bash: /tmp/.mount_FreeCAuXtklB/usr/lib/libtinfo.so.6: no version information available (required by /bin/bash)
could not open file Map.T.H for source file surfaceGenerateBoundingBox.C due to No such file or directory
could not open file UList.T.H for source file surfaceGenerateBoundingBox.C due to No such file or directory
could not open file HashSet.T.H for source file surfaceGenerateBoundingBox.C due to No such file or directory
could not open file token.T.H for source file surfaceGenerateBoundingBox.C due to No such file or directory
could not open file Map.T.H for source file surfaceToFMS.C due to No such file or directory
could not open file UList.T.H for source file surfaceToFMS.C due to No such file or directory
could not open file HashSet.T.H for source file surfaceToFMS.C due to No such file or directory
could not open file token.T.H for source file surfaceToFMS.C due to No such file or directory
```
Ubuntu 19.10

oliveroxtoby, 2020-04-23:
These errors can be ignored as they refer to files which are named differently under blueCFD. The Linux installation does not find the BlueCFD files and vice versa, but they are not needed and protected by #ifdefs. The wmake system nevertheless prints an error about them not being found. 

Let it run to the end and you should find it working.

closed 2020-04-23 by oliveroxtoby

j2l, 2020-04-23:
Thank you, I have 2 other "bugs" but I'll write my own posts.

j2l, 2020-04-23:
BTW, I'm not the OP, not sure if his case is solved.

joseRey, 2020-05-02:
I will try @oliveroxtoby 's fix.  Not as critical because snappyHexMesh works perfectly still.  Thank you very much.