jaheyns/CfdOF #34: HiSa not installed correctly in FreeCAD

sebastiaansiemons, 2020-05-26:
I tried to install HISA, and run one of the tutorials to check for any issues. None of the tutorials work because HISA was not found:
--> FOAM Warning : 
    From function bool Foam::dlLibraryTable::open(const Foam::fileName&, bool)
    in file db/dynamicLibrary/dlLibraryTable/dlLibraryTable.C at line 105
    could not load "libhisaFiniteVolume.so"

I updated HISA from the preferences tab and according to the depency checker it installed correctly:
[image]

However, the output console gives me some errors:
Checking CFD workbench dependencies...
Checking FreeCAD version
Checking for OpenFOAM:
Running  echo $WM_PROJECT_VERSION
Raw command:  ['C:/Program Files/FreeCAD 0.18/bin\\python.exe', '-u', 'C:\\Users\\MyPC\\AppData\\Roaming\\FreeCAD\\Mod\\CfdOF\\WindowsRunWrapper.py', 'C:/PROGRA~1/BLUECF~1/OPENFO~1.X\\..\\msys64\\usr\\bin\\bash', '--login', '-O', 'expand_aliases', '-c', 'echo $WM_PROJECT_VERSION']
Setting environment for OpenFOAM 5.x mingw-w64 Double Precision (of5-64), using MSMPI71 - please wait...
Environment is now ready. Notes:
   - You can change between installed versions by running: ofmenu
   - You can change to other predefined versions by running: ofmenuNew
5.x
Running  cartesianMesh -version
Raw command:  ['C:/Program Files/FreeCAD 0.18/bin\\python.exe', '-u', 'C:\\Users\\MyPC\\AppData\\Roaming\\FreeCAD\\Mod\\CfdOF\\WindowsRunWrapper.py', 'C:/PROGRA~1/BLUECF~1/OPENFO~1.X\\..\\msys64\\usr\\bin\\bash', '--login', '-O', 'expand_aliases', '-c', 'cartesianMesh -version']
Setting environment for OpenFOAM 5.x mingw-w64 Double Precision (of5-64), using MSMPI71 - please wait...
Environment is now ready. Notes:
   - You can change between installed versions by running: ofmenu
   - You can change to other predefined versions by running: ofmenuNew
/*---------------------------------------------------------------------------*\
| =========                 |                                                 |
| \\      /  F ield         | OpenFOAM: The Open Source CFD Toolbox           |
|  \\    /   O peration     | Version:  5.x                                   |
|   \\  /    A nd           | Web:      www.OpenFOAM.org                      |
|    \\/     M anipulation  |                                                 |
\*---------------------------------------------------------------------------*/
/*   Windows 32 and 64 bit porting by blueCAPE: http://www.bluecape.com.pt   *\
|  Based on Windows porting (2.0.x v4) by Symscape: http://www.symscape.com   |
\*---------------------------------------------------------------------------*/
Build  : 5.x-963176928289
Exec   : C:/PROGRA~1/BLUECF~1/ofuser-of5/platforms/mingw_w64GccDPInt32Opt/bin/cartesianMesh.exe -version
Date   : May 26 2020
Time   : 21:29:13
Host   : "MyPC"
PID    : 6992
I/O    : uncollated
Case   : C:/PROGRA~1/BLUECF~1/msys64/home/ofuser
nProcs : 1
SigFpe : Enabling floating point exception trapping (FOAM_SIGFPE).
fileModificationChecking : Monitoring run-time modified files using timeStampMaster (fileModificationSkew 10)
allowSystemOperations : Allowing user-supplied system call operations

// * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * //
cfMesh-CfdOF (cfMesh 1.1.2) 1.6
Running  hisa -version
Raw command:  ['C:/Program Files/FreeCAD 0.18/bin\\python.exe', '-u', 'C:\\Users\\MyPC\\AppData\\Roaming\\FreeCAD\\Mod\\CfdOF\\WindowsRunWrapper.py', 'C:/PROGRA~1/BLUECF~1/OPENFO~1.X\\..\\msys64\\usr\\bin\\bash', '--login', '-O', 'expand_aliases', '-c', 'hisa -version']
Setting environment for OpenFOAM 5.x mingw-w64 Double Precision (of5-64), using MSMPI71 - please wait...
Environment is now ready. Notes:
   - You can change between installed versions by running: ofmenu
   - You can change to other predefined versions by running: ofmenuNew
/*---------------------------------------------------------------------------*\
| =========                 |                                                 |
| \\      /  F ield         | OpenFOAM: The Open Source CFD Toolbox           |
|  \\    /   O peration     | Version:  5.x                                   |
|   \\  /    A nd           | Web:      www.OpenFOAM.org                      |
|    \\/     M anipulation  |                                                 |
\*---------------------------------------------------------------------------*/
/*   Windows 32 and 64 bit porting by blueCAPE: http://www.bluecape.com.pt   *\
|  Based on Windows porting (2.0.x v4) by Symscape: http://www.symscape.com   |
\*---------------------------------------------------------------------------*/
Build  : 5.x-963176928289
Exec   : C:/PROGRA~1/BLUECF~1/ofuser-of5/platforms/mingw_w64GccDPInt32Opt/bin/hisa.exe -version
Date   : May 26 2020
Time   : 21:29:15
Host   : "MyPC"
PID    : 5644
I/O    : uncollated
Case   : C:/PROGRA~1/BLUECF~1/msys64/home/ofuser
nProcs : 1
SigFpe : Enabling floating point exception trapping (FOAM_SIGFPE).
fileModificationChecking : Monitoring run-time modified files using timeStampMaster (fileModificationSkew 10)
allowSystemOperations : Allowing user-supplied system call operations

// * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * //
--> FOAM Warning : 
    From function void* Foam::dlOpen(const Foam::fileName&, bool)
    in file MSwindows.C at line 1336
    LoadLibrary failed. "Error 126: The specified module could not be found.
\
"
hisa 1.1.3
Checking for paraview:
Checking for Plot workbench:
Checking for gmsh:
Completed CFD dependency check

Any help is welcome!

oliveroxtoby, 2020-05-27:
This should be resolved by the update to HiSA. Please re-install HiSA.

closed 2020-05-27 by oliveroxtoby

sebastiaansiemons, 2020-05-27:
Yes solved!! Many thanks.