jaheyns/CfdOF #35: Hisa not found after install completed

Estay1991, 2020-06-24:
I am new to freeCAD. Just installed it today, but there is a problem with Hisa. It always says Hisa is not found after install complete.
[image]

Estay1991, 2020-06-24:
@oliveroxtoby

Estay1991, 2020-06-24:
@Titi9237, please help

Estay1991, 2020-06-24:
@oliveroxtoby

Estay1991, 2020-06-24:
[image]

oliveroxtoby, 2020-06-27:
Please could you read https://forum.freecadweb.org/viewtopic.php?f=37&t=33492&sid=3f65758ad69376c655edcaddf5d5bf0d and provide the information requested, particularly the version information from FreeCAD.

bakakouhai, 2020-07-05:
Hello @oliveroxtoby.

Since I cannot find the solution for meshing problem in 20.04 before, I want to try it in 18.04.
I encounter this problem also.

The system is running in Virtualbox.
Pop_OS 18.04
FreeCAD 18.4 installed via synaptic.
OpenFOAM 4.x with HELYX-OS

The report log is attached:
[hisa.log](https://github.com/jaheyns/CfdOF/files/4874566/hisa.log)

Kind regards,
Iwan

oliveroxtoby, 2020-07-07:
I have been able to reproduce this - for some reason BlueCFD commands are not running properly in certain versions of FreeCAD. I can't figure out what the issue is but I have now switched the default Windows installation to the OpenCFD Docker package. This seems to resolve the problem.

oliveroxtoby, 2020-07-07:
> Hello @oliveroxtoby.
>
> Since I cannot find the solution for meshing problem in 20.04 before, I want to try it in 18.04.
> I encounter this problem also.
>
> The system is running in Virtualbox.
> Pop_OS [...]

This is a different issue to the one above, but I will look into it. For now I suggest using a new version of OpenFOAM - this seems to be an issue compiling HiSA in OpenFOAM v4.

commit a28f273e (2020-07-07): Add OpenCFD Docker image as default Windows installation

Resolves #35
closed 2020-07-07 by oliveroxtoby