jaheyns/CfdOF #36: Mesh task dialog is not showing

bakakouhai, 2020-06-24:
Hello.

I am trying to do CfdOF elbow tutorial  from opensim.
However, the mesh task dialog is not showing so I cannot apply the mesh.
The workbench is installed successfully, along with cfmesh and Hisa.
Is there a workaround for this issue?

System used:
Pop_OS 20.04
OpenFOAM 7 [installed via ppa]
Freecad 18.4 [ubuntu repo via synaptic]

Kind regards,
Iwan

oliveroxtoby, 2020-06-27:
I'm not able to reproduce this problem. Please could you read https://forum.freecadweb.org/viewtopic.php?f=37&t=33492&sid=3f65758ad69376c655edcaddf5d5bf0d and supply all the information requested.

bakakouhai, 2020-06-28:
Sorry for the lack of information. here are the screenshots, logs, and elbow.fcstd.

Kind regards,
Iwan

[mesh issue.zip](https://github.com/jaheyns/CfdOF/files/4841477/mesh.issue.zip)

oliveroxtoby, 2020-06-28:
It is complaining that the 'task' panel is already open. If you click on the 'Tasks' tab in the combo view on the left, you should see it there.

closed 2020-06-28 by oliveroxtoby

bakakouhai, 2020-06-28:
@oliveroxtoby That is the problem. The task panel is empty as the screenshot inside the zip file showed.

oliveroxtoby, 2020-06-28:
Ah, ok. I see there is an error "libnglib.so: cannot open shared object file". Looks like there is a packaging error - you'll need to install whatever package provides that file on your system. Probably called netgen or something like that. This may be related to this problem: https://forum.freecadweb.org/viewtopic.php?f=37&t=46496