jaheyns/CfdOF #37: Problem compiling cfMesh-cfdof on Ubuntu 20.04

adodge, 2020-07-19:
Hello!

OS: Ubuntu 20.04

Using: OpenFOAM-7 (see https://openfoam.org)
Build: 7-3bcbaf946ae9

FreeCAD 0.18 Revision: 16146 (Git)

I've installed the "Plot" and "CfdOF" workbenches using the Addon Manager.

Clicking the "Install cfMesh" button fails, though.  I've attached the log file.

[log.Allwmake.txt](https://github.com/jaheyns/CfdOF/files/4944772/log.Allwmake.txt)

I looked at the first error, and it seemed to be looking for a directory that doesn't exist:

```
wmakeLnIncludeAll: running wmakeLnInclude on dependent libraries:
    wmakeLnInclude error: base directory /opt/openfoam7/src/edgeMesh/ does not exist
```

It looks like this has been moved to /opt/openfoam7/src/meshTools/edgeMesh/.  So I went around to all the `options` files and removed references to this directory.  `Allwclean ; Allwmake` and I get a different error, seemingly to do with syntax or typing or something.  (I don't really know C++.)

```
utilities/checkMeshDict/checkMeshDict.C:361:69: error: ‘LITERAL’ is not a member of ‘Foam::keyType’
  361 |                     dict.lookupEntry(objectNames[objectI], keyType::LITERAL);
      |                                                                     ^~~~~~~
```

[log2.txt](https://github.com/jaheyns/CfdOF/files/4944788/log2.txt)

I don't know if this is a real error that I revealed by resolving the import error, or a new error that I introduced by messing with the imports...

Does anyone have any insights?  Thanks

olesenm, 2020-07-20:
Hi @adodge 

I can't comment on possible changes to the openfoam.org version, but other items of your report and logs looks like you have a bit of a mishmash of things.

The error
```
utilities/checkMeshDict/checkMeshDict.C:361:69: error: ‘LITERAL’ is not a member of ‘Foam::keyType’
  361 |                     dict.lookupEntry(objectNames[objectI], keyType::LITERAL);
```

This `keyType` handling was added for OpenFOAM-v1812 to replace the older pair of bools that guided searching. So instead of the older code `lookupEntry(key, false, false)` you now have something more descriptive `lookupEntry(key, keyType::LITERAL)`

Your error indicates that you are using cfMesh code that is meant for compiling with the www.openfoam.com flavour, not the openfoam.org flavour (indeed quite confusing).

Furthermore, your log file has references to things like `Map.T.H`, which makes it look like the sources that additionally been adjusted for Windows compilation (special filename renaming of template files to avoid problems with the case-insensitive Windows filesystem).

If you need cfMesh on Ubuntu, try using the [ubuntu binary packs for openfoam2006 (20.06)](https://develop.openfoam.com/Development/openfoam/-/wikis/precompiled/debian). They should already include cfMesh utilities.

Note that these ubuntu packs install into `/usr/lib/openfoam` (not `/opt/openfoam`). Don't be shocked at the number of sub-packages, they have been split up to support the range of a minimal runtime env, developer support, additional tutorials.

|User Group     | Command          |
|---------------|------------------|
| minimalist    | `sudo apt-get install openfoam2006` |
| traditional   | `sudo apt-get install openfoam2006-dev` |
| everything    | `sudo apt-get install openfoam2006-default` |

There are probably more details you might need. Not sure if this will still be the right place, but I'll try to keep a watch on the issue for a bit.

oliveroxtoby, 2020-07-20:
Thanks for reporting. This should be fixed. The wmakeLnInclude errors can be ignored and relate to support for multiple versions of OpenFOAM.
@olesenm, we install a custom version of cfMesh ([https://sourceforge.net/projects/cfmesh-cfdof/]) with ifdefs to support building under both versions as well as BlueCFD. There was a mistake in one of these.

closed 2020-07-20 by oliveroxtoby

olesenm, 2020-07-20:
> There was a mistake in one of these.

Makes sense.

adodge, 2020-07-20:
Awesome!  I clicked the install button, it re-downloaded, compiled, and installed without a problem.

Thank you both!