jaheyns/CfdOF #39: Allrun FOAMDIR= set wrong

TimScottz8, 2020-08-15:
I'm running FreeCAD 0.18.4 on Ubuntu
Description:    Ubuntu 18.04.5 LTS
Release:        18.04
Codename:       bionic

I've installed CfdOF successfully and the dependency check is OK. Under the freecad preferences CfdOF the OpenFoam install directory is set to /opt/openfoam6. Meshing works absolutely fine. When I go to the Solver module and write the case out it sets the Allrun FOAMDIR="None" everytime. The work around is just to edit the Allrun in kate and set the FOAMDIR="/opt/openfoam6" and all is well but it would be nice not to have to and will probably confuse new users. It took me a while to figure out the problem being new to OpenFoam

oliveroxtoby, 2020-08-15:
I have not had any success trying to reproduce this and can't figure out why it would be OK in the mesh case and not the solver one. I have added a few lines of debugging output that gets written to the Report view when the dependency check is run. If you could update CfdOF, re-run the dependency check and send the entire output of the report view, this may help to give some clue. See here for info about the report view: https://forum.freecadweb.org/viewtopic.php?f=37&t=33492

TimScottz8, 2020-08-16:
I'll try and reproduce it later this evening Thanks for the quick replyTim Sent from my Samsung Galaxy smartphone.

TimScottz8, 2020-08-17:
I tried to reproduce the issue last night after re-installing CfdOF and couldn't. So it seems that reinstalling solved the problem :) . For anyone else who has the problem then the workaround is simple. Just manually insert the path to openfoam in the AllRun file.
Thanks for the help

closed 2020-08-17 by oliveroxtoby