jaheyns/CfdOF #40: HISA not found even after multiple installations

lokeshamh, 2020-08-29:
Hi,
I'm facing following issue when I run dependency checker in preference menu of FreeCAd, to use CFDOF, I believe that is something similar to the error that ca be find in these link: #12 , I am using OpenFoam 8, I tried installing multiple no of time but same issue, I even cloned hisa repository to Open FOAM8, still the same issue. I need your support in fixing this issue.

[image]

Thanks 
Lokesha

oliveroxtoby, 2020-08-29:
HiSA does not yet support OpenFOAM v8, but I will look into this.

lokeshamh, 2020-08-29:
Ok, meanwhile i will install OpenFOAM v7 and run it for now.

oliveroxtoby, 2020-11-11:
This has now been resolved.

closed 2020-11-11 by oliveroxtoby