jaheyns/CfdOF #43: UAV case slicemesh meshrefinement Err

skunk69, 2020-12-03:
11:12:00  Traceback (most recent call last):
  File "C:\Users\...\AppData\Roaming\FreeCAD\Mod\cfdOF\CfdMeshRefinement.py", line 133, in execute
    obj.Shape = shape
<class 'AttributeError'>: 'FeaturePython' object has no attribute 'Shape'

oliveroxtoby, 2020-12-03:
I think you may be editing a file that was created prior to a recent update of the workbench, which has broken backward compatibility of the mesh refinement objects. Please could you try removing and re-adding any mesh refinement objects?

If not, please provide a saved files and instructions on reproducing this error.

skunk69, 2020-12-03:
yes，I will try again

skunk69, 2020-12-03:
It's done, You helped. Thank you!

closed 2020-12-03 by oliveroxtoby