jaheyns/CfdOF #44: Allmesh: line 9: syntax error near unexpected token `>'

skunk69, 2020-12-09:
Allmesh: line 9: `    "$@" 1> >(tee -a log."$sol") 2> >(tee -a log."$sol" >&2)'

#!/bin/bash

runCommand()
{
    if [ "$1" == "mpirun" ]; then sol="$4"; else sol="$1"; fi
    sol=$(basename -- "$sol")
    sol="${sol%.*}"
    if [ -f log."$sol" ]; then rm log."$sol"; fi
    "$@" 1> >(tee -a log."$sol") 2> >(tee -a log."$sol" >&2)
    err=$?
    if [ ! $err -eq 0 ]; then exit $err; fi
}

# Extract feature edges
runCommand surfaceFeatureEdges -angle 60 "constant/triSurface/Compound_Geometry.stl" "Compound_Geometry.fms"
runCommand cartesianMesh

if [ ! -z $FOAM_API ] && [ $FOAM_API -ge 1906 ]
then
    runCommand surfaceMeshExtract mesh_outside.stl
else
    runCommand surfaceMeshTriangulate mesh_outside.stl
fi

runCommand surfaceTransformPoints -scale "(1000 1000 1000)" mesh_outside.stl mesh_outside.stl

oliveroxtoby, 2020-12-09:
Please provide details on how to reproduce this, prefereably from a saved file.

skunk69, 2020-12-09:
I will send you the FCstd file.

张勇
京航天程

 联系我

skunk69, 2020-12-09:
Well, I find a strange thing : I have two laptop,one thinkpad with intel core cpu and another radeon core cpu, the same windows 10, same freecad and cfdOF. but only in the thinkpad the UAV case work well

closed 2020-12-09 by skunk69