jaheyns/CfdOF #45: Support debian openfoam installation

einhander, 2021-01-25:
Hello, I've installed openfoam via `apt install openfoam`, but CfdOF can't detect openfoam. In https://develop.openfoam.com/Development/openfoam/-/wikis/precompiled/debian says that "The OpenFOAM applications are located under /usr/bin and configuration files under /usr/share/openfoam/etc", but It can't be detected in /usr/bin nor /usr/share/openfoam/ directory.

oliveroxtoby, 2021-01-25:
I don't have a debian installation to test on, so could you tell me how you run an openfoam application on your system - one of the tutorials, say? On most platforms, you have to source an openfoam bashrc file <openfoam installation dir>/etc/bashrc before you can access any of the openfoam commands. Is that the case with your installation, or are the commands accessible immediately from a terminal?

Also, which version did you install? The comment above might not be relevant to the package you installed.

einhander, 2021-01-25:
Hi, I think it's debian problem see https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=973574 
Sorry for the noice.

oliveroxtoby, 2021-01-26:
I have a feeling that the Debian installation may be integrated directly into the system instead of requiring the bashrc file to be sourced, so this may not be a problem. If you open a terminal and type an OpenFOAM command, say 'simpleFoam', does it run (and print out usage information)? Or do you just get an 'unknown command' error?

einhander, 2021-01-28:
I moved to openfoam.com ubuntu repo, but for a clearance I've installed debian version again (removed openfoam.com version first).
 
> say 'simpleFoam', does it run (and print out usage information)? Or do you just get an 'unknown command' error?

```
$ simpleFoam 

--> FOAM FATAL ERROR :

    Could not find mandatory etc entry (mode=ugo)
    'controlDict'
```
It seems that binaries form debian don't have bin folder with scripts, may be something else.

Btw, debian has openfoam version 1906.191111+dfsg1-2

oliveroxtoby, 2021-01-28:
Thank you for this information, this answers my question. It seems the debian package is integrated into the system so that the bashrc file does not have to be sourced before running commands. I will see if I can update the workbench to work with this.

einhander, 2021-01-28:
Hmm, I have the same output from openfoam from openfoam.com  repo.

einhander, 2021-01-28:
I'm trying to go further with  openfoam.com binary version and trying to install cfMesh, but build stops on following error
```
./Allwmake: 3: wmake: not found
./Allwmake: 6: wmake: not found
./Allwmake: 9: wmake: not found
```

einhander, 2021-01-28:
> but build stops on following error

Solved, by installing openfoam2012-default package from openfoam.com repo (don't know if  openfoam2012-dev package is sufficient).
Debian repo doesn't contain -dev package of openfoam.

oliveroxtoby, 2021-01-28:
I will have to request someone with a Debian system to deal with this issue, as it will need some interactive testing to figure out.

luzpaz, 2021-01-29:
kkremitzki is a Debian maintainer. Although he's pretty busy. I'd ping him as a last resort.

oliveroxtoby, 2021-01-29:
@luzpaz Thanks. There's little mystery to me about it - I'm pretty sure I know how it works, but I don't think it's going to be a good idea to develop it 'blind'. Seems like a good place to see if we can get someone else involved.

Kartstig, 2021-02-14:
I also had this issue running Ubuntu 20.04 LTS. 

I chose to symlink and source the file in my bashrc:
```
'/home/herman/.openfoamrc' -> '/opt/openfoam8/etc/bashrc'
```
and in my ~/.bashrc:
```
source ~/.openfoamrc
```

Now my output of `simpleFoam` shows:
```
herman@carnage ⏎  simpleFoam
/*---------------------------------------------------------------------------*\
  =========                 |
  \\      /  F ield         | OpenFOAM: The Open Source CFD Toolbox
   \\    /   O peration     | Website:  https://openfoam.org
    \\  /    A nd           | Version:  8
     \\/     M anipulation  |
\*---------------------------------------------------------------------------*/
Build  : 8-340defec456f
Exec   : simpleFoam
Date   : Feb 14 2021
Time   : 12:29:45
Host   : "carnage"
PID    : 320054
I/O    : uncollated
Case   : /opt/openfoam8/etc
nProcs : 1
sigFpe : Enabling floating point exception trapping (FOAM_SIGFPE).
fileModificationChecking : Monitoring run-time modified files using timeStampMaster (fileModificationSkew 10)
allowSystemOperations : Allowing user-supplied system call operations

// * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * * //
Create time

--> FOAM FATAL ERROR: 
cannot find file "/opt/openfoam8/etc/system/controlDict"

    From function virtual Foam::autoPtr<Foam::ISstream> Foam::fileOperations::uncollatedFileOperation::readStream(Foam::regIOobject&, const Foam::fileName&, const Foam::word&, bool) const
    in file global/fileOperations/uncollatedFileOperation/uncollatedFileOperation.C at line 538.

FOAM exiting

```

oliveroxtoby, 2021-02-14:
@Kartstig Please can we keep this issue to the topic of the Debian installation. Please could you open a separate issue relating to Ubuntu and describing your problem in full. I have tested on Ubuntu 20.04 and I do not experience this problem.

oliveroxtoby, 2021-02-28:
In CfdOF, leaving the installation path unset (blank) should resolve the debian problem since this should use the system-integrated OpenFOAM installation. I have not tested this but I am going to close this issue - please report if it does not work.

Islam0mar, 2021-07-03:
That doesn't work.
I have done many workarounds to make it work-for environmental variables without corrupting the program logic, then I stuck with wmake-, so it is better to use the following apt-repository:
```
sudo sh -c "wget -O - https://dl.openfoam.org/gpg.key | apt-key add -"
sudo add-apt-repository http://dl.openfoam.org/ubuntu
sudo apt install openfoam8
```
 [source](https://openfoam.org/download/8-ubuntu/)

oliveroxtoby, 2021-07-03:
@Islam0mar just for clarity, are you saying that on Debian, my proposed solution (keeping the OpenFOAM installation path blank in CfdOF) does not work for you? It would be helpful if you could provide more information about the error you encounter, so that we can try to move closer to solving the problem.

Islam0mar, 2021-07-04:
Alright. Debian package doesn't correct `bashrc` internals. I mean that bashrc assumes relative path, take for example:
 - [etc/config.sh/functions](https://github.com/OpenFOAM/OpenFOAM-8/blob/master/etc/config.sh/functions#L76)--bashrc calls it-- looks for `$WM_PROJECT_DIR/bin/foamEtcFile` which doesn't exist in Debian.  
 - any thing like $WM_PROJECT_DIR/bin or /lib doesn't work as well.
 
I stripped the package version from bashrc without sourcing and set `source_env` to `False`. That worked well for checking. I tried to compile cfmsh from freecad which doesn't work as it looks for wmake and debian doesn't include it [debian-openfoam-filelist](https://packages.debian.org/buster/amd64/openfoam/filelist) or I didn't find it, so I stopped at this point and used the mentioned apt-repository.

closed 2021-07-13 by oliveroxtoby

oliveroxtoby, 2021-07-13:
> Alright. Debian package doesn't correct `bashrc` internals. I mean that bashrc assumes relative path, take for example:
> 
> * [etc/config.sh/functions](https://github.com/OpenFOAM/OpenFOAM-8/blob/master/etc/config.sh/functions#L76)--bashrc calls it-- looks for `$WM_PROJECT_DIR/bin/foamEtcFile` which doesn't exist in Debian.
> * any thing like $WM_PROJECT_DIR/bin or /lib doesn't work as well.
> 
> I stripped the package version from bashrc without sourcing and set `source_env` to `False`. That worked well for checking. I tried to compile cfmsh from freecad which doesn't work as it looks for wmake and Debian doesn't include it [debian-openfoam-filelist](https://packages.debian.org/buster/amd64/openfoam/filelist) or I didn't find it, so I stopped at this point and used the mentioned apt-repository.
> 
> I forgot that setting `OpenFOAM installation path blank` would give an error as `bashrc` is located in `/usr/share/openfoam/etc`.
> 
> I used Ubuntu openfoam which is the same as Debian [ubuntu-openfoam-filelist](https://packages.ubuntu.com/focal/amd64/openfoam/filelist)

Thanks for the info. This helped me reproduce the issue with the package in ubuntu, and it turns out my initial assumption was wrong - as you say, you do indeed have to source the bashrc file in order to use it. It should now be working if the installation directory is set to ```/usr/share/openfoam```, although as you note, there are some issues with this package and I would agree that it's better to use the one from the openfoam website.

commit ef5aace5 (2021-07-13): Update to work with system integrated OpenFOAM package

Resolves #45