jaheyns/CfdOF #47: Preference Page is not working

jadye21, 2021-02-28:
[image]

When trying to access the CfdOF Preference page it came up blank, and at the bottom in yellow lettering it says, "Cfd Preference Page is not a preference page"

oliveroxtoby, 2021-02-28:
I'm not able to reproduce this issue. Please could you provide your version information - go to Help | About, and click "Copy to clipboard".

Which installation are you using - the system package manager's, AppImage, etc...?

jadye21, 2021-02-28:
The version information is - version 0.18 build 16093 +53 (Git)
I installed using the Ubuntu Software app.
Thanks for your help.

oliveroxtoby, 2021-03-01:
It seems that there are some issues using this with git versions less than 16146.

Please have a look at the Readme for options of installing a newer version - I would suggest the 0.19 prerelease as this is about to become the stable version.

closed 2021-03-01 by oliveroxtoby

adityahiyer, 2025-08-22:
I'm having the same issue now - I tried to post about it on the forum, but I think my post wasn't approved. I'm also getting some errors regarding the CfdTools and CfdPreferencePage Python files. 

My installation details are: 
OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/India (en_IN)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
Installed mods: 
  * AddonManager 2025.8.16
  * CfdOF 1.31.7

Not sure how to fix this.

oliveroxtoby, 2025-08-25:
Sorry, this was a bug. Should be fixed now.