jaheyns/CfdOF #50: Dependancy checker does not find GMSH

FCLC, 2021-07-30:
Running FreeCad 0.19 from the Ubuntu PPA, 
CfdOF from the manager. 
Open foam version 2012 
Gmsh latest stable from developer, pre-compiled binary, version 4.8.4

Path to GMSH is exported to bashrc and executable to all users 

Platform is PopOS 20.04 with a 5.10.x Kernel

luzpaz, 2021-07-30:
The daily PPA or the stable PPA ?

oliveroxtoby, 2021-07-30:
If the path is exported in .bashrc, it will only be available to FreeCAD if you start it from the terminal. It may be simpler to put a symlink to gmsh in somewhere that is in your global path like /usr/local/bin

closed 2021-07-30 by oliveroxtoby

FCLC, 2021-07-30:
> The daily PPA or the stable PPA ?

stable!

oliveroxtoby, 2021-08-01:
I hope my suggestion resolved your issue, but if not, please could you include the full output from the report view when the dependency checker is run?

FCLC, 2021-08-04:
> If the path is exported in .bashrc, it will only be available to FreeCAD if you start it from the terminal. It may be simpler to put a symlink to gmsh in somewhere that is in your global path like /usr/local/bin

Hi Oliver!

Out of habit I'd already been starting freecad from the terminal without any success in detecting GMSH. 

Trying your second recommendation, that did work. 

it may justify a separate issue, but I'd been having issues with paraview loading correctly with version available in the repositories and so on. Para would open, but throw an error along the lines of "could not find module paraview.py"

this would happen even if I pointed CfdOF directly to the binary 

Building the latest stable release from source and pointing CfdOF to that seems to have fixed this (I'd assume using the latest precompiled binary would work just as well)

oliveroxtoby, 2021-08-09:
Glad it's working one way or another. 

The paraview issue may be the same as #51