jaheyns/CfdOF #51: Paraview launch button not working with appimage due to path

swampyclark, 2021-08-09:
It looks like the paraview button does not work at least with the appimage due to the fact it does not explicitly attempt to launch the script in the default /tmp/meshCase folder

0.0: Preparing meshing ...
0.0: Exporting mesh refinement data ...
0.1: Exporting the part surfaces ...
0.3: Mesh case written to /tmp/meshCase
5.3: Please edit the case input files externally at: /tmp/meshCase 
0.0: Running cfMesh ...
0.0: Mesher started
1.9: Meshing completed
Reading mesh ...
Triangulated representation of the surface mesh is shown - 
Please view in Paraview for accurate display. 
0.0: Running /opt/paraviewopenfoam56/bin/paraview --script=pvScriptMesh.py
0.0: Paraview started
 
This fails to launch as does from the cli

(base) mike@mike-HP-Pavilion-Laptop-15-cs0xxx:~$ /opt/paraviewopenfoam56/bin/paraview --script=pvScriptMesh.py

explictly changing to
(base) mike@mike-HP-Pavilion-Laptop-15-cs0xxx:~$ /opt/paraviewopenfoam56/bin/paraview --script=/tmp/meshCase/pvScriptMesh.py

works fine.

I also have the same problem launching parview from the solver dialog and can work around it from the cli.

I believe the easy fix for the appimage directory issue is just use the full path  instead of expecting it to be in that directory.
Saw you had stated this was a known issue in another thread but here is a workaround.

Thanks for all the work
Mike

OS: Ubuntu 21.04 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.20.25429 (Git) AppImage
Build type: Release
Branch: master
Hash: acce57a25d82b0408fa07b9452ed31c8a06c7a1b
Python version: 3.9.6
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.2
Locale: English/United States (en_US)

[](url)

oliveroxtoby, 2021-08-09:
Thanks. This actually wasn't the a problem with the path since it is run from the case directory. It relates to some gymnastics that are done to remove the Appimage prefixes from the environment so that paraview is run as if from the system environment rather than the appimage. Anyhow, I think it is fixed now.

closed 2021-08-09 by oliveroxtoby

swampyclark, 2021-08-09:
just installed and it worked,  thanks

commit ead9502d (2021-08-09): Remove empty environment strings when stripping appimage paths

Resolves #51