jaheyns/CfdOF #52: CFD_Mesh not using multi threaded parameter for mesh case writing surface enumaration

FCLC, 2021-08-17:
Much as the title. 

Essentially, Cfd OF seems to only use the multi threaded parameter in the CFD mesh workbench when calling GMSH/snappyhexmesh or CFmesh, but does not use it when writting the case itself. 

This is fine for small meshes (number of faces smaller than 100k) but means that larger cases can take more time to write the mesh case file than actually mesh.

oliveroxtoby, 2021-08-17:
Could you elaborate - is this something that is already availble in the mesh workbench (or elsewhere) for triangulating the surface meshes, or is this a new feature request?

FCLC, 2021-08-17:
This is already available and used for both surfaces and meshes in different work benches such as Mesh Design and OpenSCAD

oliveroxtoby, 2021-08-17:
Brilliant, didn't know that! Thanks!

FCLC, 2021-08-17:
No worries! 

essentially, I'm currently using Cfd OF

FCLC, 2021-08-17:
Oops, didnt meant to close the issue!

FCLC, 2021-08-17:
Worth noting, the same issue applies when using "all faces" on a body in for selecting walls/inlet/outlet etc. tho this mostly applies to walls.

For example a very complex f1 style front to midsection with front wing and barge board can have 250k+ faces in a 2m*1m*0.5m space, and selecting "all faces" will select each one at a time (I suspect something along the lines of a while/for loop of  i in get.[number of faces] ?)

oliveroxtoby, 2021-08-18:
> This is already available and used for both surfaces and meshes in different work benches such as Mesh Design and OpenSCAD

I had a look at Mesh Design and I don't see any multithreading being used in the standard MeshPart.meshFromShape() function. gmsh supports multithreading as you say, but I don't actually see it being activated. Is this what you are suggesting using? Otherwise could you give me some specific instructions to find this feature?

oliveroxtoby, 2021-08-18:
> Worth noting, the same issue applies when using "all faces" on a body in for selecting walls/inlet/outlet etc. tho this mostly applies to walls.
>
> For example a very complex f1 style front to [...]

After a recent update it's possible to select whole objects instead of addressing faces individually. Hopefully that improves this situation...

FCLC, 2021-08-19:
> After a recent update it's possible to select whole objects instead of addressing faces individually. Hopefully that improves this situation...

Sounds good, will pull the update and take a look this evening!

FCLC, 2021-08-19:
> I had a look at Mesh Design and I don't see any multithreading being used in the standard MeshPart.meshFromShape() function. gmsh supports multithreading as you say, but I don't actually see it [...]

Not certain how the mechanism is working in that case, tho it may be that they were bulk selecting the features using the same mechanism changed above.

FCLC, 2021-08-19:
Took a quick look just now: 

the full model selection for adding an object will all faces worked perfectly! 
Would it be possible to also have a button in the UI in addition to the double click functionality? when selecting models inside other walls/other geometry can be cumbersome  to have to disable individual pieces one at a time to "dig" out the object you want to add 

Writing the mesh case still seems to select one face at a time, tho I'm not certain if that was addressed by the latest commit.

oliveroxtoby, 2021-08-24:
> the full model selection for adding an object will all faces worked perfectly!
> Would it be possible to also have a button in the UI in addition to the double click functionality? when selecting [...]

This should be possible using the "Select from list" function? Just don't choose "Select individual components" - then you can select whole objects at a time.

luzpaz, 2022-05-13:
Is this on a (soft) roadmap ? just wondering

oliveroxtoby, 2022-05-14:
> Is this on a (soft) roadmap ? just wondering

Yes - I am planning on looking at this next.

oliveroxtoby, 2023-01-02:
I have added the (experimental) functionality to do the surface meshing in parallel using gmsh. It can be activated using the 'undocumented' boolean switch called ```SurfaceTriangulationUsingGMSH``` (set it using Tools | Parameter editor, in Preferences/Mod/CfdOF).

Seems to work for me but not giving me any speed advantage over the default method.

I'm closing this now, but you are welcome to provide more detailed input or testing if you want to take this further.

closed 2023-01-02 by oliveroxtoby