jaheyns/CfdOF #55: SnappyHexMesh scaling

Serince, 2021-11-03:
Hi, 
I am running Freecad 0.19 in ubuntu machine. When I mesh with snappyhexmesh a simple model, it is scaling the mesh and the mesh model is almost double (x and y directions) of my model. I have checked for other mesh algorithims working fine. But snappyhexmesh has problem in Allmesh file I have checked thant when I change (play) the rates in command "runCommand surfaceTransformPoints -scale "(1000 1000 1000)" mesh_outside.stl mesh_outside.stl" model is getting similar sizes.

oliveroxtoby, 2021-11-03:
I have not seen this issue before - please could you upload a saved .FCStd file and instructions to reproduce the problem?

Also please state what version of OpenFOAM you are using.

Thanks!

Serince, 2021-11-03:
blockMeshDict file is wrong. The cfdof write wrong coordinate of xmin and xmax. You can check the files at the attachment.
By the way, I am using openfoam 8.
[snappy.zip](https://github.com/jaheyns/CfdOF/files/7466471/snappy.zip)

oliveroxtoby, 2021-11-03:
The problem is your "Point in mesh" is not inside the block. Use the 'Search' button to automatically pick a point inside.

closed 2021-11-03 by oliveroxtoby