jaheyns/CfdOF #56: "No module name freecad.plot" error persists after pip installing

paanduv, 2021-12-15:
I pip installed freecad.plot and it is successfully installed. Yet, this issue persists on double-clicking on cfdsolver. Do I need to take additional steps for freecad to recognize the new add-on. I have restarted freecad multiple times already. Thanks in advance!

10:47:13  Traceback (most recent call last):
  File "/home/adwaith/.FreeCAD/Mod/CfdOF/CfdSolverFoam.py", line 168, in setEdit
    from CfdRunnableFoam import CfdRunnableFoam
  File "/home/adwaith/.FreeCAD/Mod/CfdOF/CfdRunnableFoam.py", line 33, in <module>
    from CfdResidualPlot import ResidualPlot
  File "/home/adwaith/.FreeCAD/Mod/CfdOF/CfdResidualPlot.py", line 31, in <module>
    from freecad.plot import Plot
<class 'ModuleNotFoundError'>: No module named 'freecad.plot'

oliveroxtoby, 2021-12-15:
Please install the plot workbench using the addon manager, and make sure it appears in the list of workbenches. I'm not sure how you managed to do it with pip? Please see the Readme for details.

paanduv, 2021-12-16:
Thanks, fixed!

-- 
​Adwaith Gupta
CEO |
​MS | Stanford University
B.Tech | NIT Warangal

www.paanduv.com
Call at +91 8791470881
Email at ***@***.***
Follow us on LinkedIn <https://in.linkedin.com/company/paanduv-applications>
Connect with me on LinkedIn <https://www.linkedin.com/in/adwaith-gupta/>
Subscribe to YouTube
<https://www.youtube.com/channel/UCd0lfMEjSgY7Pi5C9g8xLdw>
Contact <http://paanduv.com/contact-us> Us

Paanduv Applications Private Limited is registered under the Companies Act
of India, 2013. ***@***.***, Paanduv. The content of this email is
confidential and intended for the recipient specified in the message only.
It is strictly forbidden to share any part of this message with any third
party, without a written consent of the sender. If you received this
message by mistake, please reply to this message and follow with its
deletion, so that we can ensure such a mistake does not occur in the future.

closed 2021-12-16 by oliveroxtoby