jaheyns/CfdOF #62: Installation error (missing component)

sitadrost, 2022-06-10:
I'm getting the following error while trying to install CfdOF in FreeCAD on Ubuntu 20.04 (for FreeCAD details see below):

Some errors were found that prevent to install this workbench: **Some errors were found that prevent to install this workbench.** Please install the missing components first.

So apparently I'm missing some component, but I'm not sure which one(s)....
I installed cfMesh and HiSA using the links listed in the Getting Started section. The OpenFOAM version I'm using is Foundation version 7, which comes with Paraview 5.6 (installed from source, located under /opt)

The installation instructions say something about dependency checking through FreeCAD preferences, but I can't find this option (Edit | Preferences is there alright, but I can't find any CfdOF there):
Dependencies can be checked and installed conveniently from the CfdOF Preferences panel in FreeCAD. In the FreeCAD window, select Edit | Preferences ... and choose "CfdOF".

As I couldn't find this CfdOF Preferences panel, I was also unable to set the OpenFOAM install directory, so I tried starting FreeCAD after sourcing OpenFOAM, but that didn't work either, got me the same error message.

How can I find out which component I'm missing?

Many thanks,
Sita

FreeCAD details (AppImage, FreeCAD_0.19.3-Linux-Conda_glibc2.12-x86_64.AppImage)
OS: Ubuntu 20.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.24366 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0f9259c)
Hash: 0f9259cda103ae1824ac16c68ac9b4a0d54b05fc
Python version: 3.9.7
Qt version: 5.12.9
Coin version: 4.0.0
OCC version: 7.5.3
Locale: English/United States (en_US)

oliveroxtoby, 2022-06-10:
Please have a look at the comments in this issue: https://github.com/jaheyns/CfdOF/issues/61

I suspect it's the matplotlib python package that's missing, but using version 0.20 of FreeCAD should give you much better feedback from the Addon manager.

sitadrost, 2022-06-10:
Thanks a lot, I'll give that a try.
Sorry for not spotting that issue earlier!

sitadrost, 2022-06-13:
Wow, that was surprisingly easy: CfdOF installed without any trouble using a newer version of FreeCAD (FreeCAD_weekly-builds-29126-2022-06-12-conda-Linux-x86_64-py310.AppImage). Thanks again for your help!

closed 2022-06-13 by sitadrost