jaheyns/CfdOF #64: Failed starting paraview

edombek, 2022-09-06:
Launch paraview on gui and get this error on console:
```
Traceback (most recent call last):
  File "/home/edombek/.local/share/FreeCAD/Mod/CfdOF/./CfdOF/Solve/TaskPanelCfdSolverControl.py", line 302, in openParaview
    self.open_paraview = CfdTools.startParaview(case_path, script_name, self.consoleMessage)
  File "/home/edombek/.local/share/FreeCAD/Mod/CfdOF/./CfdOF/CfdTools.py", line 1227, in startParaview
    CfdConsoleProcess.removeAppimageEnvironment(env)
AttributeError: type object 'CfdConsoleProcess' has no attribute 'removeAppimageEnvironment'
Traceback (most recent call last):
  File "/home/edombek/.local/share/FreeCAD/Mod/CfdOF/./CfdOF/Solve/TaskPanelCfdSolverControl.py", line 302, in openParaview
    self.open_paraview = CfdTools.startParaview(case_path, script_name, self.consoleMessage)
  File "/home/edombek/.local/share/FreeCAD/Mod/CfdOF/./CfdOF/CfdTools.py", line 1227, in startParaview
    CfdConsoleProcess.removeAppimageEnvironment(env)
AttributeError: type object 'CfdConsoleProcess' has no attribute 'removeAppimageEnvironment'
```

paraview 5.10.1
freecad 0.21.0.30252.gaf18a9ea3b
CfdOF latest version

oliveroxtoby, 2022-09-06:
Already fixed I think - https://forum.freecadweb.org/viewtopic.php?f=37&t=71579

closed 2022-09-06 by oliveroxtoby

edombek, 2022-09-06:
Yes, I can open manualy. But will there be a fix for opening from freecad?