jaheyns/CfdOF #72: Meshing with cfMesh with # processes > 1 causes meshing to hang.  Zombie processes left after Stop.

linuxguy123, 2023-02-08:
When meshing the attached FreeCAD project with cfMesh if number of processes is set to more than 1 and number of threads set to 0, the meshing process will get hung at "Finished renaming boundary patches" with the processors running at 100% load.   

If number of processes is set to 1 and number of threads to 0, all the processors still get used but the mesh completes without hanging. 

If one kills the mesh process with the "Stop" button, a number of zombie processes get left running, albeit at almost zero load.

[hung mesh session.txt](https://github.com/jaheyns/CfdOF/files/10680573/hung.mesh.session.txt)
[image]
[naca2412_BH.zip](https://github.com/jaheyns/CfdOF/files/10680585/naca2412_BH.zip)

oliveroxtoby, 2023-02-08:
I haven't been able to reproduce the hang on my system. Could you confirm your OpenFOAM and cfMesh version numbers?

linuxguy123, 2023-02-10:
I'll dig into this in a few days.  It is not a show stopper right now.

linuxguy123, 2023-03-19:
> I haven't been able to reproduce the hang on my system. Could you confirm your OpenFOAM and cfMesh version numbers?

This is happening on the remote machine.   

Here is the process list for the mesh that gets hung:
```
$ ps aux | grep Mesh
me        134162  0.0  0.0 174292 17336 ?        Sl   21:58   0:00 mpiexec -np 8 /home/me/OpenFOAM/me-2206/platforms/linux64GccDPInt32Opt/bin/cartesianMesh -parallel
me        134163  0.0  0.0 221328   944 ?        S    21:58   0:00 tee -a log.cartesianMesh
me        134164  0.0  0.0 221328   892 ?        S    21:58   0:00 tee -a log.cartesianMesh
me        134168  102  0.3 4034732 237604 ?      Rl   21:58   5:22 /home/me/OpenFOAM/me-2206/platforms/linux64GccDPInt32Opt/bin/cartesianMesh -parallel
me        134169  102  0.3 4038380 231280 ?      Rl   21:58   5:22 /home/me/OpenFOAM/me-2206/platforms/linux64GccDPInt32Opt/bin/cartesianMesh -parallel
me        134170  102  0.3 4029584 221532 ?      Rl   21:58   5:22 /home/me/OpenFOAM/me-2206/platforms/linux64GccDPInt32Opt/bin/cartesianMesh -parallel
me        134171  102  0.3 4063928 258864 ?      Rl   21:58   5:22 /home/me/OpenFOAM/me-2206/platforms/linux64GccDPInt32Opt/bin/cartesianMesh -parallel
me        134172  102  0.3 4035376 234968 ?      Rl   21:58   5:22 /home/me/OpenFOAM/me-2206/platforms/linux64GccDPInt32Opt/bin/cartesianMesh -parallel
me        134173  102  0.3 4032120 233276 ?      Rl   21:58   5:22 /home/me/OpenFOAM/me-2206/platforms/linux64GccDPInt32Opt/bin/cartesianMesh -parallel
me        134174  102  0.3 4030184 221800 ?      Rl   21:58   5:22 /home/me/OpenFOAM/me-2206/platforms/linux64GccDPInt32Opt/bin/cartesianMesh -parallel
me        134175  102  0.3 4066644 257484 ?      Rl   21:58   5:22 /home/me/OpenFOAM/me-2206/platforms/linux64GccDPInt32Opt/bin/cartesianMesh -parallel
```

luzpaz, 2023-03-19:
@linuxguy123 can you link to the answer ?

linuxguy123, 2023-03-19:
All I know is that it happens if you set processes to n >1 (4,6,8) while also setting threads to 1 when running cfmesh.  It is easily avoidable once this is know.