jaheyns/CfdOF #73: Excessive rounding in data entry fields

trilbytim, 2023-03-13:
Data entry fields seem to round to 2 significant figures and I can't see any way to change this behaviour. For example on Inlet velocity, if I want to set a 10 deg AoA then I need to set the Uy and Uz components to high precision, however if I enter the components to 5 sf:
[image]
 then they get rounded to just 2 sf:
[image]
Which isn't really enough. I have checked and it is not just the display that is being rounded, it's the rounded numbers which are being written into the OpenFOAM case files. So my current workaround is to edit those files with a Text Editor to enter in the numbers at full accuracy.
If I select the Inlet object in the model tree then I can edit the values to full precision in the "Data" tab, however if I then open the GUI for the inlet then it rounds them back to 2sf.

oliveroxtoby, 2023-03-13:
You can set the number of decimals in the FreeCAD preferences, in the "General" section under the "Units" tab. I think it might be tricky to get it to accept inputs with more significant figures nonetheless, though it might be possible - I'll have a bit of a look.

trilbytim, 2023-03-19:
Thank you, yes that does work, I increased the number of decimal places in that diagram. I hadn't come across this before though, entering values in dimensions in Sketcher for example I've always been able to enter as much precision as I've needed.

closed 2023-03-19 by trilbytim