jaheyns/CfdOF #77: Issues with meshing with a 2d extrusion on OpenFOAM v2206 and above

darrengarvey, 2023-03-21:
My use-case has a simple 3d box and it has a 2D planar mesh extrusion refinement. This works fine in openfoam v2106 but causes an issue in v2206 and above.

I think I narrowed down the breakage to a change in `createPatch` which I created [an issue](https://develop.openfoam.com/Development/openfoam/-/issues/2726) for, which includes some detail.

Apart from this issue, v2206 and v2212 seem to work with CfdOf. And if I use `createPatch` from v2106 and keep everything else the same things work as expected.

I wanted to raise this issue here in case this is a known issue or there is some known workaround for it.

oliveroxtoby, 2023-03-22:
Could you upload your case (.fcstd file)? I think I recall encountering this problem before and thought I had implemented a way around it. The 2D extrusion cases 'Elbow' and 'ViscousTubeBundle' in the 'Demos' directory seem to be working ok in 2206...

oliveroxtoby, 2023-05-12:
Closing for now. Feel free to re-open if this is still an issue.

closed 2023-05-12 by oliveroxtoby