jaheyns/CfdOF #87: Tidy up internal obj (turbulence) variable names to reduce confusion

oliveroxtoby, 2023-04-01:
In GitLab by @icojb25 on Mar 31, 2022, 06:50

oliveroxtoby, 2023-04-01:
In GitLab by @icojb25 on Mar 31, 2022, 06:53

Variable naming (esp for turbulence models) between the initial (field) and boundary condition classes is very confusing, and this makes it hard at times to know which variables are being used to populate which portions of the /0 directory files, especially if we are using boundary values to initialise the internal fields. This is also more important as we add more and more turbulence models, some of which share common variables (i.e. k, nut etc)

linuxguy123, 2023-04-07:
It would also be nice to have access to the constants used in the turbulence models, especially for SA.