jaheyns/CfdOF #89: Third party mesh importing

oliveroxtoby, 2023-04-01:
In GitLab by @icojb25 on Apr 16, 2022, 04:00

It seems basic mesh importing from 3rd party sources (fluent, CGNS, Star etc) is something users want, and would be useful. 

It seems at this stage, that it would be best to have either 1) mesh from part or 2) mesh import exclusive to one another, for the following reasons
1) It seems it would be quite a stretch at this stage to build in extraction of existing geometry from imported meshes firstly, and then somehow work in the ability to do mesh refinements on imported meshes. 
2) It seems importing a mesh contrary to build your own geo and meshing that, is contrary to the idea of importing an existing mesh

It seems then that Mesh Import should only be available if no existing mesh is present, and no mesh from part option should be available if a mesh import has occured. Also then that the Import mesh object will not live under the Part Mesh object.

linuxguy123, 2023-04-06:
I've actually done a bit of work on this already as part of importing meshes from the FEM workbench.

As long as OpenFOAM has a way to use the imported mesh (ie Fluent to Foam, for example) it is pretty straightforward to do EXCEPT one needs to tie the patches used in the imported mesh into what is used in CfdOF.

What I did was add another meshing type called Import Fem to the list of available meshing engines.  I then changed the Write Mesh button to Import Mesh. 

[image]

[image]

[image]

linuxguy123, 2023-04-06:
The biggest FreeCAD structural issue with importing meshes is that the meshing object should be changed to add an importing meshing engine for each kind of mesh that will be imported - FEM, Fluent, etc. 

At the same time the mesh object could be modified to use remote hosts.  As it sits right now the mesh object assumes everything is done locally.

linuxguy123, 2023-04-06:
@oliveroxtoby probably already knows this, but here is the list of mesh conversion functions available for OpenFOAM for the rest of us.

```
Mesh conversion
--
  |  
ansysToFoam | Convert an ANSYS input mesh file (exported from I-DEAS) to OpenFOAM format
 
ccmToFoam | Reads CCM files as written by PROSTAR/STARCCM and writes an OPENOpenFOAM polyMesh
 
foamToCcm | Translates OPENOpenFOAM mesh and/or results to CCM format
 
cfx4ToFoam | Convert a CFX 4 mesh to OpenFOAM format
 
datToFoam | Reads in a datToFoam mesh file and outputs a points file. Used in conjunction with blockMesh
 
fireToFoam | Convert AVL/FIRE polyhedral mesh to OpenFOAM format
 
fluent3DMeshToFoam | Convert a Fluent mesh to OpenFOAM format
 
fluentMeshToFoam | Convert a Fluent mesh to OpenFOAM format, including multiple region and region boundary handling
 
foamMeshToFluent | Write an OpenFOAM mesh in Fluent mesh format
 
 
foamToFireMesh | Write an OpenFOAM mesh in AVL/FIRE fpma format
 
foamToStarMesh | Write an OpenFOAM mesh in STARCD/PROSTAR (v4) bnd/cel/vrt format
 
foamToSurface | Extract boundaries from an OpenFOAM mesh and write in a surface format
 
gambitToFoam | Convert a GAMBIT mesh to OpenFOAM format
 
gmshToFoam | Reads .msh file as written by Gmsh
 
ideasUnvToFoam | I-Deas unv format mesh conversion
 
kivaToFoam | Convert a KIVA grid to OpenFOAM
 
mshToFoam | Convert .msh file generated by the Adventure system
 
netgenNeutralToFoam | Convert a neutral file format (Netgen v4.4) to OpenFOAM
 
plot3dToFoam | Plot3d mesh (ascii/formatted format) converter
 
 
star4ToFoam | Convert a STARCD/PROSTAR (v4) mesh into OpenFOAM format
 
tetgenToFoam | Convert tetgen .ele and .node and .face files to an OpenFOAM mesh
 
vtkUnstructuredToFoam | Convert legacy VTK file (ascii) containing an unstructured grid to an OpenFOAM mesh without boundary information
 
writeMeshObj | For mesh debugging: writes mesh as three separate OBJ files
```

https://www.openfoam.com/documentation/user-guide/a-reference/a.2-standard-utilities