jaheyns/CfdOF #93: Add screenshot to each subdirectory within Demos

oliveroxtoby, 2023-04-01:
In GitLab by @david.yoga on Sep 6, 2022, 15:04

It would be excellent if the in each `Demos/../README.md` there would be an attached screenshot of the demo itself (i.e. before and after images)

oliveroxtoby, 2023-04-01:
In GitLab by @oliveroxtoby on Sep 6, 2022, 19:52

I guess that would be helpful. I'm looking forward to your pull request! :P

icojb25, 2024-03-12:
@oliveroxtoby  I guess there are a couple more images to add, not covered in the PR above, so we should keep this open for the meantime

luzpaz, 2025-08-08:
what's left here ?

oliveroxtoby, 2025-09-16:
Thanks for looking at this @muezabdalla. CFD is by nature quite demanding on mesh resolution and I feel these meshes are already close to the limit of how coarse they can be without becoming totally implausible, so I would prefer not to go that route. I'm afriad 4GB is a rather modest amount of memory, especially for CFD applications.

closed 2025-10-21 by oliveroxtoby