jaheyns/CfdOF #141: CfdOF not responding. Running demos and simple model.

C-monC, 2023-04-25:
Hi,

On ubuntu 22
I have installed the FreeCad 0.20.2 conda AppImage.
Using installed hisa and cfmesh (not docker but I have tried the docker image).
OpenFoam version 9 from apt install openfoam9.

The error:
[image]

It seems like the simulation does complete. After reopening freecad after killing it the solution in paraview returns this error.

[image]

I initially tried the demos and assumed they were out of date but I have made a basic rectangular tube example that also results in FreeCad not responding.

Dependancy checker output:

> Checking dependencies...
> System: Linux
> Runtime: Posix
> OpenFOAM directory: /opt/openfoam9
> Checking dependencies...
> System: Linux
> Runtime: Posix
> OpenFOAM directory: /opt/openfoam9
> Paraview executable: /opt/paraviewopenfoam56/bin/paraview
> gmsh executable: /usr/bin/gmsh
> Completed CFD dependency check
> Paraview executable: /opt/paraviewopenfoam56/bin/paraview
> gmsh executable: /usr/bin/gmsh
> Completed CFD dependency check
> Checking dependencies...
> System: Linux
> Runtime: Posix
> OpenFOAM directory: /opt/openfoam9
> Paraview executable: /opt/paraviewopenfoam56/bin/paraview
> gmsh executable: /usr/bin/gmsh
> Completed CFD dependency check

I can confirm that all the demos also freeze FreeCAD.
Memory and CPU usage remain low and I have left the simulation for several hours.

I have attached the FCStd as a zip.
[example_freeze.zip](https://github.com/jaheyns/CfdOF/files/11324759/example_freeze.zip)

C-monC, 2023-04-27:
I compiled FreeCad, reinstalled openFoam 9 and I'm using the docker image but I still have the CfdOF freeze right after the simulation completes.

Manually running the simulation works. 
```
cd /tmp/case
simpleFoam # not necessary if I started the sim in freecad

paraview pv.foam
```
[image]

I have also just noticed this error/warning for Qprocess which pops up when the UI freezes.

[image]

Do you have any ideas as to where this issue originates from (a specific file/line in this repo)?

C-monC, 2023-04-27:
I see it doesn't necessarily finish the simulation. The UAV demo doesn't complete.

> Starting time loop
> 
> Time = 1
> 
> smoothSolver:  Solving for Ux, Initial residual = 0.13908732, Final residual = 0.0074376413, No Iterations 1
> smoothSolver:  Solving for Uy, Initial residual = 0.049523763, Final residual = 0.00094335403, No Iterations 2
> smoothSolver:  Solving for Uz, Initial residual = 0.063253364, Final residual = 0.00092508066, No Iterations 2
> GAMG:  Solving for p, Initial residual = 1, Final residual = 0.0053147058, No Iterations 6
> GAMG:  Solving for p, Initial residual = 0.0033777953, Final residual = 2.3825864e-05, No Iterations 5
> GAMG:  Solving for p, Initial residual = 0.00064757502, Final residual = 5.5164133e-06, No Iterations 5
> GAMG:  Solving for p, Initial residual = 0.00014836018, Final residual = 1.3436998e-06, No Iterations 5
> GAMG:  Solving for p, Initial residual = 4.7266477e-05, Final residual = 4.2853621e-07, No Iterations 5
> GAMG:  Solving for p, Initial residual = 1.4899016e-05, Final residual = 1.4532247e-07, No Iterations 5
> time step continuity errors : sum local = 7.3976029e-07, global = -5.4583983e-07, cumulative = -5.4583983e-07
> smoothSolver:  Solving for omega, Initial residual = 7.5984468e-05, Final residual = 4.1378296e-06, No Iterations 3
> smoothSolver:  Solving for k, Initial residual = 1, Final residual = 0.080416255, No Iterations 3
> ExecutionTime = 4.27 s  ClockTime = 5 s
> 
> Time = 2
> 
> smoothSolver:  Solving for Ux, Initial residual = 0.10961021, Final residual = 0.0049485325, No Iterations 1
> smoothSolver:  Solving for Uy, Initial residual = 0.10004515, Final residual = 0.0053341389, No Iterations 1
> smoothSolver:  Solving for Uz, Initial residual = 0.065559269, Final residual = 0.0024366527, No Iterations 1
> GAMG:  Solving for p, Initial residual = 0.18728523, Final residual = 0.0012793835, No Iterations 5

The last console output before it dies.

C-monC, 2023-05-03:
I changed the output directory to my home directory but FreeCad still hangs after a while.
I'll try on a different ubuntu pc when I get a chance.

C-monC, 2023-05-03:
I tried changing the directory to my home directory but Freecad still freezes.
I think I'll just use Cfodf to generate the case folder and manually run the simpleFoam in there for now.

oliveroxtoby, 2023-05-03:
Yeah, sorry...no idea at this stage. Really hard to debug this sort of issue without being able to reproduce it. Please let me know if you find out anything further.

oliveroxtoby, 2023-05-12:
@C-monC a similar issue was described here on Windows: https://forum.freecad.org/viewtopic.php?t=78096
Workaround was to disable routing python output to the report view. It would be great if you could check whether this avoids your issue as well.

oliveroxtoby, 2024-10-22:
Similar error reported here: https://forum.freecad.org/viewtopic.php?p=787876#p787876

NothanUmber, 2024-11-15:
Thanks @oliveroxtoby  for linking this my thread from the FreeCAD forums. Yes, disabling routing python output to the report view indeed fixed the problem for me, too.
My system information:
Using FreeCAD 1.0 (RC2 back then, now RC4), Windows 11 23H2 (staying away from 24H2 as long as I still can in order to keep my MixedReality headset running), Nvidia driver 565.90 (RTX 3080 10 GB VRAM), AMD Ryzen 5000gen CPU, 64 GB RAM.