jaheyns/CfdOF #143: Steady-state MultiPhase Sim

finnBsch, 2023-07-08:
Hi,
I am trying to get a steadystate solution in a multiphase simulation. I know that interfoam does not support real steady state solutions, however according to this https://openfoam.org/release/2-0-0/steady-state-vof/ , the _local-time stepping_ method can get good results for pseudo steady state solution. Is there a way to get something similar using CfdOF?

sstevenson638, 2023-07-10:
You may be able to edit the OpenFOAM files generated by cfdOF. I'm a beginner so I don't know exactly what to do either. The cfdOF forum is a good place to ask this question. 
https://forum.freecad.org/viewforum.php?f=37

oliveroxtoby, 2023-07-11:
CfdOF doesn't support this, but as @sstevenson638 says, you can always use it to generate a starting case for further modification.

closed 2023-07-11 by oliveroxtoby