jaheyns/CfdOF #146: Feature request:  Solver for heat transfer

Supermagnum, 2023-08-02:
Case example:
This heat exhanger pulls heat out of a cooling loop that flows 20 liters per minute, input water temperature is 33 degrees cesius, the heat exhangers material is aluminium. The loop itself isn't modeled, but it is made from 6x   4 meter long tubes in parallel, each tube has i diameter of 10mm. Total volume of cooling liquid: 72 liters.
The tubes are PVC  10mmØ ID x 13mmØ OD 
 It uses 4X 50W peltier elements to pull heat out of the heat exhanger, into a perfect infinite heatsink. Elapsed time is 150 seconds. 
What is the cooling loops liquid temperature after end of elapsed time ?
Ambient air temperature is 40 degrees celsius.
https://github.com/Supermagnum/heatsink

luzpaz, 2023-08-16:
Ping @oliveroxtoby

oliveroxtoby, 2023-08-18:
Thanks. This has been requested a few times and will do into the queue for possible future development.

Just be advised that I am unfortunately unlikely to take this on as it's difficult for me to justify working for free on major new features.

Supermagnum, 2023-08-18:
> Sorry, I deleted my previous reply as I read this in a hurry, saw the word 'heatsink' and thought you were asking for a CHT solver ;-)
> 
> I presume a 'perfect infinite heatsink' could be modelled by a fixed-temperature boundary condition, and we already have a heat-transfer solver, which is selected by unticking 'isothermal' in the 'Physics' task panel. Does this sound like it would work for you? I'm sorry I haven't read all the details, so I'm not sure if this will be sufficient, but basically you can solve flow with heat transfer in a single region. Any heat transfer from elsewhere has to be handled through boundary conditions.

Perhaps,  but there also are a heat exhanger that pulls 1.8 KW out of 3 kilos of ice.
It is made from 4 meters of aluminium tube.
It is used to cool the liquid in a cooling loop, that liquid is heated by 100W. 
All the calculations for how much ice is needed is in the github repository, but those doesn't include efficiency of the ice to coolant heat exhanger, loss to ambient air and so on..