jaheyns/CfdOF #148: QoL Improvement: Make CfdAnalysis' "Output Directory" dynamic based on the Project Name, instead of "/tmp/case" as default

eduncan911, 2023-09-07:
CfdOF WB v1.24.4
FreeCAD 0.21.1 (and LinkStage Tip, which is 0.21)

**Problem:**
The current function of the Properties for CfdOF allows you to manually specify a folder for writing meshes.  However, if you are analyzing several different projects, this becomes a problem with data being constantly overwritten.

**Suggested Resolution**
This request is to allow project variables to be used here instead.  Such as the Project Label (name).  For example:

* Project is named "cfdof.example1"
* CfdOF WB Settings could have "/tmp" like it does now.
* Default would be the open root tree's "Label", aka Project Name.

[image]

**Work Around**
Currently, this can be set manually from with each project's CfdAnalysis's Property window on a per project basis.  And that works.  However, most user's workflows in FreeCAD doesn't include, "New project, go edit Property Settings before starting."  So this is kind out of the normal workflow for end-users.

[image]

oliveroxtoby, 2023-09-14:
This is a good idea. I'll try to put it in soon, or you're welcome to submit a pull request.

eduncan911, 2023-09-15:
@oliveroxtoby nice!  as a follow-up, I dove deep into FEM recently.  And, it has everything I mentioned above as well as "smarter than me" logic with even more options.

Maybe follow this Workbench?

[image]

eduncan911, 2023-09-15:
There's also something else... the FEM workbench also defaults to "search in known binary directories" for all the dependencies.

Just some suggestions to improve the the user's experience.  :)

[image]

eduncan911, 2023-09-15:
> ... you're welcome to submit a pull request.

I don't have the slightly clue where to start on that PR.  lol

icojb25, 2024-03-15:
hi @oliveroxtoby any idea what the call is to get the project base `label` is? Is this accessible via the `obj` passed to the CfdAnalysis.py class? 

`def makeCfdAnalysis(name):`
     `""" Create a Cfd Analysis group object """`
    `obj = FreeCAD.ActiveDocument.addObject("App::DocumentObjectGroupPython", name)`
    `CfdAnalysis(obj)`

oliveroxtoby, 2024-03-16:
> hi @oliveroxtoby any idea what the call is to get the project base `label` is? Is this accessible via the `obj` passed to the CfdAnalysis.py class?

FreeCAD.ActiveDocument.Label ?

oliveroxtoby, 2024-03-17:
> But note the other two dynamic ones. I especially like the "beside FCStd file" option.
@icojb25 just a note in case you end up implementing this aspect of it. CfdOF does currently store files in this location if the user enters a relative path in the analysis object properties - see CfdTools.getOutputPath(). So probably this code can be reused.

oliveroxtoby, 2024-03-17:
> There's also something else... the FEM workbench also defaults to "search in known binary directories" for all the dependencies, instead of requiring their location.
> 
> This makes the FEM workbench work out of the box with CAE Linux - since everything is installed. For CfdOF, I have to go find each binary and specify them.

CfdOF does actually have default locations for the dependencies, which are searched if no location is given on the Preferences page. However I don't know what these are for CAE Linux, so if you could supply these we can simply add them to the defaults.

icojb25, 2024-03-17:
> @icojb25 just a note in case you end up implementing this aspect of it. CfdOF does currently store files in this location if the user enters a relative path in the analysis object properties - see [...]

Hi @oliveroxtoby i can chip away at it over the week. Do we want to add these options in the Preferences panel?

oliveroxtoby, 2024-03-18:
> Hi @oliveroxtoby i can chip away at it over the week. Do we want to add these options in the Preferences panel?

Thank you, that would be great.I don't think we need to have the distinction between custom directory and tmp directories (I think we can just have the custom directory defaulting to tmp directory as we do now), but adding an option in preferences like the 'Beside .FCStd file' one would be nice. And then also saving in a subdirectory with the name of the document label, as the first post alludes to.

oliveroxtoby-engys, 2024-03-26:
> hi @oliveroxtoby I am nearly finished this implementation, however notice that in `CfdPreferences.py`, the function `loadSettings()` is actually never called, which is kind of weird. One would assume this is used to persist the changes (not working for this Feature Request yet, assume because of this).
> 
It's not called by CfdOF but it is called by FreeCAD when the preferences page is loaded.

oliveroxtoby, 2024-03-26:
> hi @oliveroxtoby I am nearly finished this implementation, however notice that in `CfdPreferences.py`, the function `loadSettings()` is actually never called, which is kind of weird. One would [...]
It's not called by CfdOF but it is called by FreeCAD when the preferences page is loaded.

icojb25, 2024-03-29:
@eduncan911 @oliveroxtoby Take a look at this and see if it works as intended. Enjoy the long weekend.

closed 2024-06-27 by oliveroxtoby

eduncan911, 2024-06-28:
> @eduncan911 @oliveroxtoby Take a look at this and see if it works as intended. Enjoy the long weekend.

Apologies for the delay.  Im between projects, so will check as soon as the next one comes!

commit 050d66b6 (2024-06-27): Add the option to save output in a subdirectory with the document name

Resolves #148
Resolves #162