jaheyns/CfdOF #150: Install Win10 CfdOF 1.24.8 OpenFoam v2212

CLGH, 2023-11-11:
After installation 2 problems
- MPI not found
- Hisa not found.

MPI solved this way : 
create folder <OpenFOAM folder>\msys64\home\ofuser\OpenFOAM\OpenFOAM-v2212\platforms\win64MingwDPInt32Opt\lib\msmpi
copied these files from MPI windows installation :
mpiexec.exe
mpitrace.man
msmpilaunchsvc.exe
smpd.exe

Hisa solved installing v2306 : includes more files than v2212 especially hisa.exe
Is hisa v2306 compatible with v2212 ?

Thanks,
Claude.

oliveroxtoby, 2023-11-13:
> Hisa solved installing v2306 : includes more files than v2212 especially hisa.exe Is hisa v2306 compatible with v2212 ?
> 

Sorry about that. It should be resolved, this was a mistake with the 2212 installation.

I will look into the MPI issue.

oliveroxtoby, 2023-11-13:
For me, the MPI issue is resolved as long as one restarts FreeCAD after installing MPI. I have added a message prompting the user to do this after they install OpenFOAM, and a note in the readme that if OpenFOAM is not installed with administrator privileges (as the code prompts one to do) then it is necessary to manually install MSMPI. Hope this works the same for you; let me know if not.

commit e7d769ff (2023-11-13): Add note about MSMPI installation

Resolves #150
commit 831ae343 (2023-11-13): Add message to restart FreeCAD after OpenFOAM installation

Resolves #150
closed 2023-11-13 by oliveroxtoby