jaheyns/CfdOF #152: Bug reporting

Azazzeli, 2023-11-26:
We ask that you please use this interface to contribute constructively to this free, open-source project by reporting carefully tested problems that you find in the software. Please provide as much information as you can and, if at all possible,
**include a small test case which allows developers to reproduce the issue**.
**
[image]
[image]
[image]
[image]

Azazzeli, 2023-11-26:
Вот такая ошибка при попытке установить дополнение

oliveroxtoby, 2023-12-26:
I'm unfortunately unable to reproduce this error, so I will need more information. Please see the forum post mentioned in the issue template for more information. It does look like it may be an issue that is more related to the FreeCAD addon manager than CfdOF, so it might be worth seeking help there.

oliveroxtoby, 2024-03-12:
Closing due to lack of response. Feel free to reopen.

closed 2024-03-12 by oliveroxtoby