jaheyns/CfdOF #156: Add-on install not working

wisehart, 2024-02-22:
I tried to install the CfdOF Add-On on a fresh install of FC. First in the Add-On manager the add-on preview page looks like this:

[image]

Then, when clicking "Install", this window appears and nothing happens:

[image]

After waiting for a long time and clicking "Cancel" this message appears in the previe window:

`Laden der README-Datei von URL https://github.com/jaheyns/CfdOF/blob/master/README.md ist fehlgeschlagen`

Python Console shows this output:

```
Python 3.8.17 (default, Jul 27 2023, 21:57:58) [MSC v.1900 64 bit (AMD64)] on win32
Type 'help', 'copyright', 'credits' or 'license' for more information.
>>> # Gui.runCommand('Std_Workbench',7)
>>> # Gui.runCommand('Std_ViewStatusBar',1)
>>> # Gui.runCommand('Std_AddonMgr',0)
>>> # Gui.runCommand('Std_AddonMgr',0)
>>>
```

Version Information:

OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git)
Build type: Release
Branch: releases/FreeCAD-0-21
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)

oliveroxtoby, 2024-03-12:
I think that these are issues with the FreeCAD addon manager rather than with CfdOF per se. I would ask you to try it with the latest development build as I suspect they have been addressed, however unfortunately CfdOF isn't yet compatible with it as there are some changes to the material system. So I will check back on this once those changes have been made.

Kai2510, 2024-04-09:
The addon manager is sometimes unstable. I chose to install it manually by cloning this repository into $FreeCAD_ROOT$/Mod directory because of bad network condition. Need to update it manually.

oliveroxtoby, 2024-07-08:
@Kai2510 could you check if the issue is resolved if you use the lastest FreeCAD development version? CfdOF is now compatible with it.

oliveroxtoby, 2024-08-16:
Closing due to lack of response.

closed 2024-08-16 by oliveroxtoby