jaheyns/CfdOF #160: Eulerian multiphase

icojb25, 2024-03-23:
@oliveroxtoby I have been doing some work on E-E multiphase lately and was thinking of putting a PR together for something like this, for the workbench?  There are quite a few different solvers to choose from, but we have the VoF stuff already set up. 

Perhaps it might be a nice addition ... 

Any thoughts and or advice / preferences?

oliveroxtoby, 2024-03-26:
Yup, thanks. Knock yourself out! I think the later versions of OF Foundation have consolidated around reactingMultiphaseEulerFoam's I/O format so might be worth looking at that as a base. But I'm not sure so do go ahead as you see fit.

icojb25, 2024-03-29:
see #163

icojb25, 2024-03-29:
@oliveroxtoby If one considers `reactingMultiphaseEulerFoam`, it seems there are a lot of options and opportunities. I wonder if we should limit this to some kind of graduated introduction. (ie i believe you can have `single phase` reactions as well as `multiphase` in OF, so how do we want to go about this? 

Perhaps we implement `reactions` for just `reactingMultiphaseEulerFoam` first (only), and then expand things. Or, should we stick to straightforward `non-reacting` Eulerian MF to begin with ... ?

icojb25, 2024-03-29:
Actually, @oliveroxtoby it looks like a lot of these separate solvers have been amalgamated into `multiphaseEulerFoam`, so I guess we go just with this at the moment. Then we can look at `buoyantReactingFoam` and all the rest.

oliveroxtoby, 2024-03-29:
Yes, I definitely wouldn't actually add any reacting flow at this stage. multiphaseEulerFoam is complex enough as it is. 

Sorry, my comment was a bit misleading, what I meant was that the new consolidated multiphaseEulerFoam in Foundation 11 is, I think, based around the case format of the reactingMultiphaseEulerFoam in earlier Foundation versions and in the OpenCFD versions. I wasn't intending to suggest you actually support reactions.

icojb25, 2024-03-29:
> Yes, I definitely wouldn't actually add any reacting flow at this stage. multiphaseEulerFoam is complex enough as it is.
>
> Sorry, my comment was a bit misleading, what I meant was that the new [...]

Hi @oliveroxtoby ok ... i guess i have implemented the basics of simple reactions in `multiphaseEulerFoam` . It seems quite tractable - do you want to take a look first?

icojb25, 2024-03-31:
A further comment here @oliveroxtoby , I agree with the above actually. The reactions part is going to need a lot of additional stuff to allow for the configurations and reaction definitions, unless we just used some default stuff. 

On the basic Eulerian stuff though,  did you have any thoughts about how configuration for the dispersed phase and interaction could be done? There are again a lot of options. To me, it seems the best option is to add a `Phases` or `Phase Physics` or `Eulerian Physics` option on the menu, and require this before writing the Case file (throw an error if its missing, similar to how a error is thrown if VOF is selected and there i only x1 fluid object). 

There is too much required to try to jam this into the existing `Physics Selection` object ... imho

oliveroxtoby, 2024-03-31:
> There is too much required to try to jam this into the existing `Physics Selection` object ... imho

Yes, I don't think the right place for this is 'Physics Selection'. I would suggest putting phase-specific settings (e.g. diameter model) into the material properties specification for each phase. I would also put inter-phase properties under the material properties, but under an additional selection of the 'other phase'. (That would also be the home for surface tension for VOF - it can't be specified at all currently.)

icojb25, 2024-03-31:
> Yes, I don't think the right place for this is 'Physics Selection'. I would suggest putting phase-specific settings (e.g. diameter model) into the material properties specification for each phase. 

Ok. Nice idea, that works well.

>I would also put inter-phase properties under the material properties, but under an additional selection of the 'other phase'.  (That would also be the home for surface tension for VOF - it can't be specified at all currently.)

Ok, I'll see what I can figure out ... so for multiphase, the user specifies at least x1 additional Material, which then populates interphase properties in the bottom of the current Material Properties object? Something like that?

For  specifying the `lift`, `drag`, and other forces etc, I have created an `Phase physics model` object (similar to the Dynamic Mesh objects) which fits quite nicely under the main `Physics Model` object. It seems to work quite nicely and looks pretty good!