jaheyns/CfdOF #161: Cfdof/FreeCAD 0.21.2 Generates Incorrect Periodic B.C.'s

hooshsim, 2024-03-23:
Huge thank you to the development team for implementing periodic boundary conditions in the latest version of FreeCAD/CfdOF...a job well done!

To take advantage of the periodic boundary conditions, I built a sector model of an axial fan installed in a wind-tunnel. In my model, I assigned P1 as slave and P2 as master surfaces using rotational option. Then I updated the fan patch in the boundary field file (0/U) for rotationality. I ran the case for about 300 iterations without any issues. The predicted moments were much smaller than expected ones. After checking the 0/U file, I realized that slave surface (P1) is missing from the patch list in the boundary field file as shown below.

I also changed p2_slave to P1_slave and re-ran the case and no improvements in the solution. Any suggestions or recommendations?

P2_master
{
type cyclicAMI;
}

P2_slave
{
type cyclicAMI;
}

Thank you

hooshsim, 2024-03-23:
Couldn't upload the case file due to unsupported file type.

oliveroxtoby, 2024-03-26:
> Couldn't upload the case file due to unsupported file type.

Could you try zipping it or uploading it on the forum please?

hooshsim, 2024-03-26:
Thanks Oliver.  See the attached zip file.  

[Fan1.zip](https://github.com/jaheyns/CfdOF/files/14764214/Fan1.zip)

oliveroxtoby-engys, 2024-03-28:
This is actually the expected behaviour. The master and slave patches are named as <master name>_master and <master name>_slave - so the P1 patch is not missing, it is just renamed to P2_slave. It's just a matter of naming so should have no bearing on results.

oliveroxtoby, 2024-03-28:
This is actually the expected behaviour. The master and slave patches are named as _master and _slave - so the P1 patch is not missing, it is just renamed to P2_slave. It's just a matter of naming so should have no bearing on results.

closed 2024-03-28 by oliveroxtoby