jaheyns/CfdOF #166: Allrun script points to incorrect MESHDIR

ben-kenney, 2024-04-10:
Hi! I just installed cfdof on Windows 11 with docker (podman) for the openfoam solver.

I noticed that my `Allrun` script sets `MESHDIR="/tmp/../meshCase"` which results in the error: 

`Fatal error: Unable to find mesh in directory /tmp/../meshCase`. 

When writing the mesh case, cfdof will write the mesh to the default output directory, not into `/tmp`

Should `Allrun` pull from the default output directory that is set in global cfdof preferences? In my case I set this to `\\wsl$\podman-machine-default\home\user\cfdof`.

In any case, an easy workaround is to manually modify the `Allrun` script to `MESHDIR="../meshCase"` which then works.

I did find [this forum post](https://forum.freecad.org/viewtopic.php?t=63352) about the same error message although it wasn't clear to me how it was resolved, maybe order of operations matters?

oliveroxtoby, 2024-04-11:
I have fixed a change I recently made that probably caused this. Could you check if it's resolved?

ben-kenney, 2024-04-11:
Yes it's working now, thanks!

Now I see that the /tmp dir is the location of the `meshCase` and `case` directories from inside the docker container.

closed 2024-04-11 by ben-kenney