jaheyns/CfdOF #174: Errors with Running the Solver

RaysKindaCrazy, 2024-06-19:
I have CfdOF v1.25.17 , OpenFOAM 11 v2212 , and running a mesh with cfMesh

I have double checked the directories, and have reinstalled incase that is the problem. But when I run the solver and re-mesh case I get the error

[image]

When I run without re-meshing the case it says

[image]

I dont see how to fix the issue, without editing the software, and I have not seen anyone post about this. If anyone has any suggestions of what I might've missed I'd love to hear since I've tried all I know

oliveroxtoby, 2024-06-26:
This seems more likely to be a support query than a bug report. Please post on the forum at https://forum.freecad.org/viewforum.php?f=37, making sure to include your FreeCAD file so that other users can attempt to reproduce your issue.

closed 2024-06-26 by oliveroxtoby

RaysKindaCrazy, 2024-06-27:
Yes, sorry and thank you

oliveroxtoby, 2024-06-29:
@RaysKindaCrazy FYI this should be resolved now