jaheyns/CfdOF #177: Feature request - Adding other solver support

mactone, 2024-08-07:
First of all, I would like to thank @jaheyns for great job on making the CfdOF. 
I would like to propose feature request. 
Since I am a combustion engineer, I am dealing with reactingFoam mostly. 
I wonder if you could add another solver support or instruct me on how to add new solver support?

Though the reactingFoam solving on CfdOf can be done by Tech Bernd's youtube channel, Freecad CfdOf Tutorial #11: Simple combustion with reactingFoam. But it will be great to have native support. 

Expecting to have this feature. 
Thank you.

oliveroxtoby, 2024-08-16:
As much as I would like to, I'm afraid this is not something I am going to have time to do for free.

Sorry about that!

oliveroxtoby, 2025-08-02:
Hi @njelich. Thank you for the offer. I think this will be a fairly big undertaking, so haven't really had a chance to sit down and think in any detail about what would be needed. An interface to define the chemical reactions would be one thing to start with. Do feel free to take a stab at it and I'll be happy to help answer any questions.

njelich, 2025-08-09:
Reacting foam.