jaheyns/CfdOF #179: Feature requests: Moving Ground and Symmetry Conditions, Please

BrendaEM, 2024-08-18:
Hi,

Firstly, thank you for creating and maintaining CFDOf . OpenFoam is a great solver, but setting up anything larger than an academic case has been...interesting. CFDOf makes it routine to set up and run a case. I just got done showing an aerodynamcist/math person at San Jose State University how great CFDOf is.

I have been using CFD to work on a car body. I am setting the surrounding walls as slip, but I think that the ground should be moving, as in x,y, and z velocity. As far as I know, OpenFoam will support this. Could moving ground please be added to boundary conditions?
Ref, I think: https://www.openfoam.com/documentation/guides/latest/doc/guide-bcs-wall-translating-wall-velocity.html

Also, there is a symmetry plane, which might be useful, too.
https://www.openfoam.com/documentation/user-guide/4-mesh-generation-and-conversion/4.2-boundaries

Thank you for your consideration--and CFDOF

P.S. Will there be another version, soon? : )

oliveroxtoby, 2024-08-19:
> Firstly, thank you for creating and maintaining CFDOf . OpenFoam is a great solver, but setting up anything larger than an academic case has been...interesting. CFDOf makes it routine to set up and [...]

Thanks for your interest. Please do report any specific issues that present obstacles in setting up larger cases.

> I have been using CFD to work on a car body. I am setting the surrounding walls as slip, but I think that the ground should be moving, as in x,y, and z velocity. As far as I know, OpenFoam will [...]

Indeed, and you can access this boundary condition it CfdOF by choosing Boundary type 'Wall' and Sub-type 'Translating'. Have you tried this?

> Also, there is a symmetry plane, which might be useful, too. https://www.openfoam.com/documentation/user-guide/4-mesh-generation-and-conversion/4.2-boundaries

Symmetry can be specified by choosing Boundary type 'Constraint' and Sub-type 'Symmetry'.

> P.S. Will there be another version, soon? : )

I try to keep refining it on an ongoing basis, and of course it is open for anyone to contribute. If you have specific requests, please submit them and I'll always give them consideration.

BrendaEM, 2024-08-22:
I was so excited to read about the translating wall (ground) that I rushed back to do a test. It's running now.
Thank you.