jaheyns/CfdOF #180: Feature request: Spinning wheels

BrendaEM, 2024-08-22:
It would be wonderful to be able to select 2 to 4 wheels on a vehicle and have them spin.
(I have noticed that CfdOf does do a single spinning thing, like windmills and prop.)

Thank you (again) for CfdOf.

[Yes, I know that the curl-like flow on the side of the wheel would be pretty boring, but the air being flung under the wheelwell might be more interesting.

I realize that, oddly, doing vehicle design seems to be minority, whereas windmills, props, cityscapes, HVAC, valves, and academic studies such as the Forward-facing-step seems quite popular, and yet I persist in trying to make an efficient vehicle.]

oliveroxtoby, 2025-06-24:
Is the 'rotating wall' boundary condition sufficient for this purpose, or are you looking for something more detailed, e.g. to handle grooves or spokes etc?

oliveroxtoby, 2025-06-25:
> 3.) Set each wheel to a rotating mesh boundary , given a specific centroid point, and some kind of angular velocity, such as surface speed, or wheel RPM.

It sounds like a rotating velocity BC for each wheel is what you want (you set axis and rotational velocity for each). This is a fairly recent addition and might not have been available when you previously tried.

> (gulp) I wish we had a coefficient-of-drag readout, by hull. It's not easy in Paraview, though this might be a better OpenFoam request. I am not sure.

Have a look at the 'Reporting function' in CfdOF - I think that will do what you are referring to.

> 
> That stated, thank you for CfdOf : )
> 
> [I've been working on a light vehicle body, with it. In the past, I have edited OpenFoam cases, but the meshing has been the difficult part without a GUI, well for things more complicated than academic cases. CfdOf has let me do a study in an evening. Thank you.]

Glad you've found it helpful!

oliveroxtoby, 2025-08-25:
Closing, as this is hopefully resolved as per the comment above.

closed 2025-08-25 by oliveroxtoby

BrendaEM, 2025-09-01:
Hi @oliveroxtoby , I am sorry that I was I have posted back.

[...Because of Windows Recall, I have moved my main system from Windows to Linux, on two systems. MS changed their installer, which caused problems. So, I have just about got it straightened out. I also needed solutions to replace Drobox, for 
anything but cat pictures.]

I have installed Freecad, Paraview. I hate Docker with all of my heart, meaning that I want Docker off of my world. Anyway, I am also having issues that CFDOf, does not seem to like the Mint distro Open Foam, so I need to RTFM, to get that working.

Anyway, thank you for continuing to work on CFDOf, as it makes OpenFoam practical, for larger cases than just small academic ones. I have set up a case or two. It's not fun, and with meshing, doing it in a text editor is impractical. 

Thanks Again,
BrendaEM