jaheyns/CfdOF #182: Bug reporting - "Check Mesh" Function Generates an Error

mas985, 2024-08-28:
Reference:

https://forum.freecad.org/viewtopic.php?t=90094

After creating the mesh and clicking on "Check Mesh" I get this error:
Code: [Select all](https://forum.freecad.org/viewtopic.php?t=90094#)

```
0.0: Mesh check started ...
0.6: OpenFOAM fatal error: cannot find file "C:/Program Files/ESI-OpenCFD/OpenFOAM/v2212/msys64/home/ofuser/OpenFOAM/system/controlDict" 
0.6: Mesh check OK
Code: [Select all](https://forum.freecad.org/viewtopic.php?t=90094#)

```
```
--> FOAM FATAL ERROR: (openfoam-2212)
cannot find file "C:/Program Files/ESI-OpenCFD/OpenFOAM/v2212/msys64/home/ofuser/OpenFOAM/system/controlDict"

    From virtual Foam::autoPtr<Foam::ISstream> Foam::fileOperations::uncollatedFileOperation::readStream(Foam::regIOobject&, const Foam::fileName&, const Foam::word&, bool) const
    in file global/fileOperations/uncollatedFileOperation/uncollatedFileOperation.C at line 561.

FOAM exiting

10:26:43  
```
It says "Mesh OK" but I don't think it actually ran

oliveroxtoby, 2024-08-29:
Thanks. Should be working now.

closed 2024-08-29 by oliveroxtoby

mas985, 2024-08-29:
Thanks, it is working now.

However, there is a new issue. After meshing and I try to run the solver, a message pops up saying I need to remesh. However, when I look at the variable "Needs Mesh Rerun" under "CfdAnalysis", it is set to false so it should not request a remesh, correct?

oliveroxtoby, 2024-08-30:
If 'Needs Mesh Rewrite' is true it will also suggest a remesh. I would guess that's the reason. However, I'm not sure why it is being set to true if you have changed nothing in the model between meshing and running the solver. I'm not able to reproduce that behaviour unfortunately.

mas985, 2024-08-30:
I have rewritten the mesh, re-created the mesh and 'Needs Mesh Rewrite' is still set to true.

mas985, 2024-08-30:
I just tried it on another older project that I know had worked before and now 'Needs Mesh Rewrite' is not being cleared after mesh creation.

mas985, 2024-08-31:
Sorry I keep finding these errors but this one occurred with Check mesh while using SnappyHexMesh

```
07:51:47  

--> FOAM FATAL IO ERROR: (openfoam-2212)
Entry 'maxNonOrtho' not found in dictionary "D:/CFD/Cases/Fittings/Pipe Only/meshCase_SHM/system/meshQualityDict"

file: system/meshQualityDict

    From bool Foam::dictionary::readEntry(const Foam::word&, T&, Foam::keyType::option, Foam::IOobjectOption::readOption) const [with T = double]
    in file /home/mol/openfoam/OpenFOAM-com/src/OpenFOAM/lnInclude/dictionaryTemplates.C at line 327.

FOAM exiting
```
Seems a variable is missing

oliveroxtoby, 2024-09-02:
> Are you on the same "versions" as I am?

Pretty much, yes. I'm not sure why this is happening for you and not for me, but I will keep it in mind.

oliveroxtoby, 2024-09-02:
> Sorry I keep finding these errors but this one occurred with Check Mesh after creating a mesh with SnappyHexMesh.

Should be fixed now

mas985, 2024-09-02:
I don't know if this matters or not but when I did a dependency check, I got the following:

`Checking dependencies...
FreeCAD version: 0.21
System: Windows
Runtime: MinGW
OpenFOAM directory: C:\Program Files\ESI-OpenCFD\OpenFOAM\v2212
OpenFOAM version: 2212
**cfMesh (CfdOF version) not found**
HiSA version: 1.9.6
Paraview executable: C:/Program Files/ParaView 5.10.1-Windows-Python3.9-msvc2017-AMD64/bin/paraview.exe
gmsh executable: C:\Program Files\FreeCAD 0.21\bin\gmsh.exe
gmsh version: 4.11.0
Completed CFD dependency check
`
Notice the cfMesh part

mas985, 2024-09-02:
Looking through some of the code, I found this routine in CfdMesh.py:

` def updateData(self, obj, prop):
        analysis_obj = CfdTools.getParentAnalysisObject(obj)
        num_refinement_objs = len(CfdTools.getMeshRefinementObjs(obj))
        num_dyn_refinement_objs = (0 if CfdTools.getDynamicMeshAdaptation(obj) is None else 1)
        if prop == "Group":
            if analysis_obj and not analysis_obj.Proxy.loading:
                if num_refinement_objs != self.num_refinement_objs:
                    analysis_obj.NeedsMeshRewrite = True
            if analysis_obj and not analysis_obj.Proxy.loading:
                if num_dyn_refinement_objs != self.num_dyn_refinement_objs:
                    analysis_obj.NeedsCaseRewrite = True
            self.num_refinement_objs = num_refinement_objs
            self.num_dyn_refinement_objs = num_dyn_refinement_objs
        else:
            if analysis_obj and not analysis_obj.Proxy.loading:
                analysis_obj.NeedsMeshRewrite = True`

I also noticed that the created mesh is not part of any group so could that be why NeedsMeshRewrite is being set to true?

[image]

commit 333b0d31 (2024-08-29): Fix issue running checkMesh under Windows

Resolves #182
commit 37b28739 (2024-09-02): Fix issue running checkMesh with snappyHexMesh of OF2312

Resolves #182