jaheyns/CfdOF #183: Bug reporting

kamakiri1225, 2024-09-04:
We ask that you please use this interface to contribute constructively to this free, open-source project by reporting carefully tested problems that you find in the software. Please provide as much information as you can and, if at all possible,
**include a small test case which allows developers to reproduce the issue**.

[image]

The dam break demo is not working.
I believe the issue lies with the InitialisationZone.
When I click on InitialisationZone, I get the following screen.
Is this the expected behavior?

[image]

In this YouTube video, the demo is working correctly. However, in the latest version of cfdOF, it does not behave as shown in the video.

https://www.youtube.com/watch?v=w7ZKDwps1p4&list=PL9H9jQE7y0a5jhlyACRzsdfnx-42AYCCX&index=15

oliveroxtoby, 2024-09-04:
I can't reproduce this behaviour - it seems to be working fine for me. I note that your initialisationZone appears to be outside the CfdAnalysis object, so your file may be broken in some way. It's hard to say without the .FCStd file.

If you would like me to investigate further, please provide the saved state as a FCStd file, and **exact** instructions on how to reproduce the problem.

kamakiri1225, 2024-09-04:
Thank you for your response. It worked correctly with FreeCAD 0.21.2. Could this be an issue of version compatibility? For now, it seems like everything will work without any problems. Thank you.

[image]

oliveroxtoby, 2024-09-04:
It could be. I will close this since v0.20 won't be supported for much longer anyway.

closed 2024-09-04 by oliveroxtoby